stm32duino/Arduino_Core_STM32 #2121: The frequency of  PWM is Wrong with STM32G030

sunnyguhz, 2023-09-06:
**Describe the bug**
I find a bug with STM32G030 series. 
The frequency of  PWM is 1.5 times, when I changed "generic_clock.c" to set SysClockFreq below 16M.
I used STM32G030C8T6 and STM32G030K6T6, all them has this problem.

**To Reproduce**
```
void setup() {
  //analogWriteResolution(12);
  analogWriteFrequency(3000); 
  pinMode(PB9, OUTPUT);
  analogWrite(PB9, 64);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

**Desktop (please complete the following information):**
 - OS: [ Windows10]
 - Arduino IDE version: [2.2.1]
 - STM32 core version: [2.6.0]
 - Tools menu settings if not the default: [Debug:None, Optimize:Smallest(-Os) with LTO, USB:None, U(S)ART:Disable]
 - Upload method: [SWD]

**Board (please complete the following information):**
 - Name: [Generic G030C8Tx]
 - Hardware Revision: [I don't know]

fpistm, 2023-09-06:
What are your changes in generic_clock.c?

fpistm, 2023-09-12:
Hi @sunnyguhz 
I've made some test and can confirm the PWM frequency is not correct.
After some investigation, it is due to a hardware issue with STM32G0xx series.
Looking at the errata sheet:
https://www.st.com/resource/en/errata_sheet/es0486-stm32g030x6x8-device-errata-stmicroelectronics.pdf
[image]

PB9 is linked to  TIM17_CH1, that's why you get this result. Using another pins not linked to TIM16 or TIM17 will work.
Unfortunately, I will not be able to provide a fix for this.

sunnyguhz, 2023-09-12:
Okay. However, it's still working.

I changed some like this in generic_clock.c
It is working in 4M now.
```
#if defined(ARDUINO_GENERIC_G030C6TX) || defined(ARDUINO_GENERIC_G030C8TX)
#include "pins_arduino.h"

/**
  * @brief  System Clock Configuration
  * @param  None
  * @retval None
  */
WEAK void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {};

  /* Configure the main internal regulator output voltage */
  HAL_PWREx_ControlVoltageScaling(PWR_REGULATOR_VOLTAGE_SCALE2);
  /*
   * Initializes the RCC Oscillators according to the specified parameters
   * in the RCC_OscInitTypeDef structure.
   */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSIDiv = RCC_HSI_DIV4;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_OFF;
  //RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  //RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  //RCC_OscInitStruct.PLL.PLLM = RCC_PLLM_DIV1;
  //RCC_OscInitStruct.PLL.PLLN = 2;
  //RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  //RCC_OscInitStruct.PLL.PLLR = RCC_PLLR_DIV2;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    Error_Handler();
  }
  /* Initializes the CPU, AHB and APB buses clocks */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_PCLK1;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSI;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_2) != HAL_OK) {
    Error_Handler();
  }
}

#endif /* ARDUINO_GENERIC_* */

```

fpistm, 2023-09-12:
Yes it works because SYSCLK is equal to PCLK:

closed 2023-09-12 by fpistm