stm32duino/Arduino_Core_STM32 #2129: Linker error after using "std::chrono::steady_clock::now()"

github-rkaiser, 2023-09-13:
If a am using

```
#  include <chrono>
...
std::chrono::steady_clock::now()

```
with a H745ZI, C031C6 or L476RG, I get the linker error

```
ld.exe: m-none-eabi\lib\thumb\v6-m\nofp\libc_nano.a(libc_a-gettimeofdayr.o): in function _gettimeofday_r
gettimeofdayr.c*: (.text._gettimeofday_r+0xe): undefined reference to _gettimeofday
 
collect2.exe*: error: ld returned 1 exit status

Error linking for board Nucleo-64 (STMicroelectronics_Nucleo_64)
Debug build failed for project 'STM32-C031C6' 
```

Please add he missing function to the library.

Thanks
Richard

fpistm, 2023-09-14:
Hi @github-rkaiser 
Thanks for this issue.
Just a note, STM32 core is not a library :wink: 

About your issue, `_gettimeofday` required to be implemented.
Example to get the build successfull you can add this to your sketch:
```
extern "C" int _gettimeofday( struct timeval *tv, void *tzvp )
{
    return 0;  // return non-zero for error
}
```

Now, which implementation could be done:
- based on the number of milliseconds passed since the board began running the current program (millis());
- based on RTC and the epoch time which will required to have STM32RTC library dependency.

fpistm, 2023-09-19:
@github-rkaiser 
any though on my previous comment?

PR #2151 (2023-10-26): feat: implement weaked _gettimeofday
Fixes #2129.

### Sketch used to validate:

#include <chrono>

```C++
void setup() {
  Serial.begin(115200);
  auto timepoint = std::chrono::steady_clock::now();
  Serial.println(std::chrono::duration_cast<std::chrono::milliseconds>(timepoint.time_since_epoch()).count());
}

void loop() {

  auto timepoint = std::chrono::steady_clock::now();
  Serial.println(std::chrono::duration_cast<std::chrono::milliseconds>(timepoint.time_since_epoch()).count());
  delay(1111);
}

```
commit 81583c57 (2023-10-26): feat: implement weaked _gettimeofday

Fixes #2129.
closed 2023-10-26 by fpistm