stm32duino/Arduino_Core_STM32 #2132: STM32 L476RG only 32 KB for upload

github-rkaiser, 2023-09-18:
After successfully compiling my  STM32 L476RG program with the Visual Micro Extension for the Arduino IDE

```
Program STM32-L476RG size: 21352 bytes (used 2% of a 1048576 byte maximum) (2,00 secs)
Minimum Memory Usage: 1420 bytes (1% of a 98304 byte maximum)

``` 
I get the message

```
Uploading 'STM32-L476RG' to 'Nucleo-64 (STMicroelectronics_Nucleo_64)' using 'COM10'
Uploader started for board Nucleo-64 (STMicroelectronics_Nucleo_64)
Upload method will be: bootloader
Uploading via Bootloader
ls\2.2.1\win\massStorageCopy.bat -I o_64\Release/STM32-L476RG.ino.bin" -O "NODE_L476RG"
Nicht genügend Speicherplatz auf dem Datenträger im aktuellen Laufwerk.
Nicht genügend Speicherplatz auf dem Datenträger im aktuellen Laufwerk.

The uploader returned an error
```
 
This sounds strange since the L476RG has capacity of 1 MB.

In the windows filesystem there is a drive  NOD_L476RG with a capacity of 32 KB.

What do I have to do to upload a program to this board?

Thanks
Richard

fpistm, 2023-09-18:
Hi @github-rkaiser 
Try to update the stlink firmware of your board.

fpistm, 2023-09-19:
Hi @github-rkaiser 
I've tested and it works. This issue is linked to the STLink firmware which handle the mass storage feature.
So as stated you have to update the STLink firmware of the board using the STM32CubeProgrammer.
Blue button "Firmware upgrade"
[image]

[image]

I close this issue as it is not related to an issue with the core.

closed 2023-09-19 by fpistm