stm32duino/Arduino_Core_STM32 #2138: STM32H5 internal compiler error

dojyorin, 2023-09-23:
# Details
Compilation fails with "internal compiler error: Segmentation fault" during compilation.

# Repeatability
- OS: Windows 11 x64
- Arduino Version: Arduino-CLI 0.34.2
- Use from: VSCode Arduino Extension
- STM32duino Version: 2.6.0
- Target Board: Nucleo-H563ZI

The problem occurs even if I compile an blank project.

```c++
void setup(){}
void loop(){}
```

# Cause
After some research, I was able to compile successfully by commenting out this part.
 https://github.com/stm32duino/Arduino_Core_STM32/blob/1ee0a03609d0a5a8dc89f792263765df1f3d1404/system/Drivers/STM32H5xx_HAL_Driver/Src/stm32h5xx_hal_dma_ex.c#L4122-L4125

# Log
<details>
<summary>ErrorLog</summary>

```log
[Starting] Verifying sketch 'arduino\sketches\template\template.ino'
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
during GIMPLE pass: evrp
C:\Users\*\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0\libraries\SrcWrapper\src\HAL\stm32yyxx_hal_dma_ex.c: In function 'HAL_DMAEx_List_ReplaceNode_Head':
C:\Users\*\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.6.0\libraries\SrcWrapper\src\HAL\stm32yyxx_hal_dma_ex.c:34:9: internal compiler error: Segmentation fault
34 | #pragma GCC diagnostic pop
|         ^~~
libbacktrace could not find executable to open
Please submit a full bug report, with preprocessed source (by using -freport-bug).
See <https://gcc.gnu.org/bugs/> for instructions.
Error during build: exit status 1
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Ctrl+Alt+I"
[Error] Verifying sketch 'arduino\sketches\template\template.ino': Exit with code=1
```
</details>

fpistm, 2023-09-25:
Hi  @dojyorin 
I was not able to reproduce on Windows 10 nor with Linux.
Anyway, it seems already reported to STMicroelectronics GH:
https://github.com/STMicroelectronics/stm32u5xx_hal_driver/issues/5
Reading this issue, you will be redirected to some bugzilla.
It is linked to gcc:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=108340

So I close it as not related to the core and will be fixed within next arm none eabi gcc toolchain update.

closed 2023-09-25 by fpistm

fpistm, 2023-09-25:
Maybe this issue could interest @ilg-ul. At least for information.

ilg-ul, 2023-09-25:
Thank you @fpistm, I'll keep in mind that there is an issue, but there is not much we can do, we'll wait for Arm to make a new release.

fpistm, 2023-09-25:
> Thank you @fpistm, I'll keep in mind that there is an issue, but there is not much we can do, we'll wait for Arm to make a new release.

Welcome, as stated it is more FYI than expecting support. I know you can't do nothing :wink: