stm32duino/Arduino_Core_STM32 #2143: Mikroelektronika Mini M4 STM32 issues

ITstreet1, 2023-10-13:
Hi,

I have a board from the title. It uses STM32F415RG MCU with a pre-programmed HID bootloader.
https://www.mikroe.com/mini-stm32f4

While using a HID bootloader method to upload a sketch over the USB, it throws an error that the IDE can not detect the board. It does not show up on the COM port. Tried with the Reset button, but no luck.
What could be a problem? Maybe the onboard bootloader is not compatible with this STM32 support?

I can always program it with a UART adapter using Arduino IDE, but thought it should be better this way.

The board works like a charm using Necto Studio.

fpistm, 2023-10-13:
The USB CDC have to be enabled to allow restart in HID bootloader.  You have to flash it the first time with USB enabled then it should be OK. Else probably it is not compatible.

ITstreet1, 2023-10-13:
Tried all options for USB support. All I get is 

> Sketch uses 26228 bytes (2%) of program storage space. Maximum is 1048576 bytes.
> Global variables use 5036 bytes (3%) of dynamic memory, leaving 126036 bytes for local variables. Maximum is 131072 bytes.
> 
> +-----------------------------------------------------------------------+
> |         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
> |     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com   |
> |     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr      |
> |   Customized for STM32duino ecosystem   https://www.stm32duino.com    |
> +-----------------------------------------------------------------------+
> 
> > Trying to open the [COM9]...
> > Unable to open the [COM9]
> > Searching for [1209:BEBA] device...
> ##########
> Error - [1209:BEBA] device is not found :(> Searching for [COM9] ...
> > Finish
> the selected serial port > Finish
>  does not exist or your board is not connected
> 

And this no matter if I try to enter the bootloader with a button or not.

ITstreet1, 2023-10-13:
There is nothing on COM9, nor is anything selected as a COM PORT option.

```
void setup() {
  Serial.begin(9600);
  pinMode(PC12, OUTPUT);
}
void loop() {
  digitalWrite(PC12, HIGH);
  delay(1000);
  digitalWrite(PC12, LOW);
  delay(1000);
}
```
I get the same message as above. For USB support tried both CDC options. Nothing. Do I have to enable USART or something else?