stm32duino/Arduino_Core_STM32 #2147: Support for H755ZI

Stelath, 2023-10-18:
**Is your feature request/improvement related to a problem? Please describe.**

Hi, I work with a team of University students working on rocketry and we cant use our H755ZI boards with Arduino core because it isn't supported, and would love if y'all could add support.

**Describe the solution you'd like**

Addition of the H755ZI Processor to the STM32 Arduino Core

**Describe alternatives you've considered**

Purchasing H753ZI boards that are supported, but we already have around 20 H755ZI Boards

fpistm, 2023-10-19:
Hi @Stelath 
I guess you ask support of the https://www.st.com/en/evaluation-tools/nucleo-h755zi-q.html ?
Did you evaluate to add them yourself? It is a community project any help are welcome.
Added to #722

fpistm, 2023-10-19:
https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-%28board%29

fpistm, 2023-11-07:
I close it as it is added to #722.

closed 2023-11-07 by fpistm

fpistm, 2023-11-07:
> How do I generate the `variant_NUCLEO_STM32H755ZI.h` and `.cpp` files? I've done the rest with the generic processor

You have to create them manually.

Stelath, 2023-11-07:
Any tips on where to find most of the info? Or do I just have to look at pinouts?

fpistm, 2023-11-07:
Most of the info of the board are in the user manual: https://www.st.com/resource/en/user_manual/um2408-stm32h7-nucleo144-boards-mb1363-stmicroelectronics.pdf

Else yes most of the work is to do proper pin mappings  and define all dedicated pins (LED, button, default Serial instance, ...) and add a clock config probabmy using the HSE.

fpistm, 2023-11-07:
Do not hesitate to create a PR with the first part of your work this will help to support you.