stm32duino/Arduino_Core_STM32 #2150: analogWrite not working on PA7 and PA8 simultaneously on STM32F103C8

DreiDe, 2023-10-24:
When unsing the `analogWrite()` function on a custom board with `Generic STM32F1 series -> BluePill F103C8` or `Generic STM32F1 series -> Generic F103C8Tx` it is not possible to generate pulses on pin `PA7` and `PA8` at the same time. The pin that `analogWrite()` is first run with produces PWM pulses successfully, the second doesn't.

So this example only produces PWM signals on `PA8`:
```cpp
void setup() {
  pinMode(PA7, OUTPUT);
  pinMode(PA8, OUTPUT);

  analogWrite(PA8, 200);
  analogWrite(PA7, 200);
}

void loop() {
}
```

When I call `analogWrite()` with `PA7` first, then PWM signals are only output on `PA7`, not `PA8`.

At first I thought this might be a hardware/timer limitation because all other pin combinations work as expected, but when I tested the same pins with MapleCore, I realized that it works without problems there. So I believe that it is a issue with the Arduino_Core.

DreiDe, 2023-10-24:
I think the problem is that the Arduino Core only allows the PA7 PWM to run on Timer1 Channel1, while according to the datasheet it should also be able to run on Timer3 Channel2. As Timer1 Channel1 is also used by PA8 there is a collision.

```cpp
//works
  HardwareTimer *Tim1 = new HardwareTimer(TIM1);
  Tim1->setPWM(1, PA7, 1000, 10);

//does not work
  HardwareTimer *Tim3 = new HardwareTimer(TIM3);
  Tim3->setPWM(2, PA7, 1000, 10);
```

fpistm, 2023-10-25:
Hi @DreiDe 
In fact you have access to all possibilities of a pin:
https://github.com/stm32duino/Arduino_Core_STM32/blob/b4d7bee9373ebf357084fa8f95cf24596184a8e0/variants/STM32F1xx/F103C8T_F103CB(T-U)/PeripheralPins.c#L97-L100

In your case, you're right, PA7 and PA8 use the same TIM1 CH1(N).
So you can simply do:
```
  analogWrite(PA8, 200);
  analogWrite(PA7_ALT1, 200);
```

`_ALTx` allows to access alternatives of a pin.

closed 2023-10-25 by fpistm