stm32duino/Arduino_Core_STM32 #2159: support for debug/release builds in Arduino IDE 2.x

m-mcgowan, 2023-10-31:
Arduino IDE 2.x supports debugging, and with that, has a new menu option *Sketch > Optimize for Debugging*. 

This looks like a convenient way to enable debugging builds with symbols from within the IDE and provides a standard means of compiling a debug build with the CLI.

The `Optimize for Debugging` option sets the `compiler.optimization_flags` placeholder from either `compiler.optimization_flags.debug` or `compiler.optimization_flags.release` according to the setting chosen. (The [debugging section](https://arduino.github.io/arduino-cli/0.34/platform-specification/#sketch-debugging-configuration) in the platform spec explains this more fully.)

`boards.txt` currently defines menus for the optimization level, and whether symbols and/or core logs are compiled in. The overlap in functionality between these existing menu items and the new "Optimize for Debugging" may need to be resolved.

As a starting point `platform.txt` would have two new placeholders defined, corresponding to the flags used for debug and release builds:

```
compiler.optimization_flags.debug=-g
compiler.optimization_flags.release=-DNDEBUG
```
The build environment then sets `compiler.optimization_flags` from either of these two, which we could propagate via `compiler.extra_flags`:

```
compiler.extra_flags=-mcpu={build.mcu} {compiler.optimization_flags} {...etc...}
```

With this in place, the Debug setting always includes debug symbols, and the Release setting does not include system logs, irrespective of the setting of the `Debug symbols and core logs` menu item.  This seems to be the least element of surprise and does "the right thing" for debug and release builds, although it may not be obvious to users that the `Optimize for Debugging` menu selection takes precedence over the `Debug symbols and core logs` menu selection.

Alternatively, the `compiler.optimization_flags.release` value could be left empty, which would mean that a release build uses the selection from `Debug symbols and core logs` exclusively, while the debug build always contains symbols, plus core logs when selected.

The latter approach seems somewhat "dangerous" in that it puts the onus on the user to be certain to deselect `core logs` in a release build.

If I were asked to make the choice, I would lean more towards the first approach with release builds explicitly excluding core logs, simply because it feels more in line with what a developer would expect for a release build, but that's just my casual suggestion. The community here may see other alternatives, or have good reasons to prefer the second approach, which I look forward to learning about and discussing.

Once we've established the most desirable way forward, I would be happy to make the changes needed and submit a PR.

m-mcgowan, 2023-10-31:
If we implement via the `{compiler.optimization_flags}` placeholder then this will be silently ignored by Arduino IDE 1.x and older versions of the CLI. (I believe undefined placeholders evaluate to the empty string, but I can verify that when testing.). 

A third option I didn't mention that occurs to me now is to **make no change** - leave things as they are. This could be acceptable if we document that the "Optimize for Debugging" option does nothing. This wasn't immediately clear to me when I started experimenting with the debug feature.

I can appreciate adding the new Optimize for Debugging option with the existing menus is tricky with the limitations of `boards.txt` :-) I'd be happy to do some experimentation to help find out what is workable/possible, such as investigating if we can have multiple optimization flags ordered so that the "Optimize for Debugging" setting takes precedence. 

Although debugging is still a "beta" feature, I've been using the debugger quite extensively lately and it seems pretty stable!

fpistm, 2023-11-15:
Hi @m-mcgowan 
I've made a PR to add this support. Tested with Arduino IDE 1.x and 2.x.

PR #2184 (2023-11-15): feat: support compiler.optimization_flags for Arduino IDE 2.X
Fixes #2159

See https://arduino.github.io/arduino-cli/0.34/platform-specification/#sketch-debugging-configuration
commit d21f8295 (2023-11-15): feat: support compiler.optimization_flags for Arduino IDE 2.X

Fixes #2159
closed 2023-11-15 by fpistm