stm32duino/Arduino_Core_STM32 #2167: Serial Read not working for STM32F030K6TX Based Custom Board

rikilshah, 2023-11-04:
```
int USART1_DE = PA1;
int USART1_TX = PA2;
int USART1_RX = PA3;

char temp;

void setup() {
  Serial.setRx(PA3);
  Serial.setTx(PA2);
  Serial.begin(9600);
  HAL_Delay(2000);
  pinMode(USART1_DE, OUTPUT);
  digitalWrite(USART1_DE, HIGH);
  Serial.println("Hello");
  delay(50);
  digitalWrite(USART1_DE, LOW);
}

void loop() {
  
  if(Serial.available())
  {
    digitalWrite(USART1_DE, HIGH);
    Serial.println(Serial.read());
    digitalWrite(USART1_DE, LOW);
  }
}

```

I am trying to perform Serial Communication on a custom board based on STM32F030K6TX. I am using USB to RS485 converter for the communication in half-duplex mode. For starter, I tried running a routine pingback code to check hardware. It successfully sends data but not able to receive data. I have tested the board and setup using STM32 HAL based code and I am able to send and receive data successfully, stating board and setup is working. 

As per my observation, Serial.available() never changes. What could be the reason? 

Again, The board is confirmed to work with STM32 HAL based code.

fpistm, 2023-11-04:
Hi @rikilshah 
Your code is not correct. Serial is not configured for half duplex and you don't have to manage extra pin. See https://github.com/stm32duino/Arduino_Core_STM32/wiki/API#enable-half-duplex-mode
Please avoid to open issue for support request. Use the forum or the GitHub discussion.

closed 2023-11-04 by fpistm

rikilshah, 2023-11-04:
Well, In this case,  USART is not supposed to be in Half Duplex mode, It is RS485 which should be in Half duplex which different than UART's mode. I am able to run exact same  logic in STM32 HAL based code.

fpistm, 2023-11-06:
Hi  @rikilshah 
I've a lot of input like this and often the same problem, it is a wrong usage. Moreover as stated, this kind of support request should be raised as a discussion or on the forum.  So sorry if I misunderstood but your code works correctly for USART connected to another USART. (used a Nucleo where USART is connected to STLink which act as USART to USB converter. Don't know why it does not work in your case. Maybe related to the hardware configuration, compare the Arduino configuration VS the HAL code you made.

rikilshah, 2024-03-16:
I finally found the issues. It was related to assertion time required in RS485 converter. 

So basically, you have to provide some delay after sending data to assert RS485 in receive mode. 

```
int USART1_DE = PA1;
int USART1_TX = PA2;
int USART1_RX = PA3;

char temp;

void setup() {
  Serial.setRx(PA3);
  Serial.setTx(PA2);
  Serial.begin(9600);
  pinMode(USART1_DE, OUTPUT);
  digitalWrite(USART1_DE, HIGH);
  Serial.println("Hello");
  delay(50);
  digitalWrite(USART1_DE, LOW);
}

void loop() {
  
  if(Serial.available() > 0)
  {
   temp = Serial.read();
   digitalWrite(USART1_DE,HIGH);
    Serial.println(temp);
    delay(50);
    digitalWrite(USART1_DE, LOW);
  }
}
```

fpistm, 2024-03-16:
Fine. Thanks for the feedback.