stm32duino/Arduino_Core_STM32 #2173: EEPROM "looses" data after core update 2.5.0 to 2.6.0.

bobemoe, 2023-11-10:
**Describe the bug**
EEPROM "looses" data after core update from 2.5.0 to 2.6.0.
This issue means it is imposable to preserve config stored in EEPROM if we update our firmware to the latest core.

**To Reproduce**
Build the following against the two different cores, or with the accompanying bash script.

firmware.ino

```c++
#include <EEPROM.h>
#include "version.h"

void setup() {
  Serial.begin(9600);
  while (!Serial) {;}

  Serial.println(VERSION);
  Serial.println("Read:");
  eeprom_buffer_fill();
  for(int address=0;address<256;address++){
      Serial.print(eeprom_buffered_read_byte(address));
      Serial.print(", ");
  }
  Serial.println();

  Serial.println("Write:");
  eeprom_buffered_write_byte(0,VERSION);
  for(int address=1;address<256;address++){
      eeprom_buffered_write_byte(address,address);
  }
  eeprom_buffer_flush();
  Serial.println("Done.");
}

void loop() {
}
```

build.sh
```bash
#!/bin/bash
arduino-cli core install STMicroelectronics:stm32@2.5.0 || exit
echo "#define VERSION 25" > version.h
arduino-cli compile -e -b STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F411CE,rtlib=nanofps,xserial=none,usb=CDCgen || exit
mv build/STMicroelectronics.stm32.GenF4/*.ino.bin eeprom_test25.bin

arduino-cli core install STMicroelectronics:stm32@2.6.0 || exit
echo "#define VERSION 26" > version.h
arduino-cli compile -e -b STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F411CE,rtlib=nanofps,xserial=none,usb=CDCgen || exit
mv build/STMicroelectronics.stm32.GenF4/*.ino.bin eeprom_test26.bin
```
You should now have 2 versions `eeprom_test25.bin` and `eeprom_test26.bin`

Steps to reproduce the behavior:
1. Preform a FULL ERASE of a STM32F11 `STM32_Programmer.sh -c port=USB1 -e all`
2. Flash eeprom_test25.bin `STM32_Programmer.sh -c port=USB1 -d eeprom_test25.bin 0x08000000 -g`
3. View serial `screen /dev/ttyACM0 9600`
```
25
Read:
255, 255, 255....
Write:
Done.
```

This shows its version number (25) and a dump of the first 256 bytes of EEPROM. Should all be 255 as the EEPROM was full just erased. Next it writes its version number to byte 0 and then fills the remaining bytes with the address 1,2,3...etc
4. Reset the STM. screen will close.
5. View serial again `screen /dev/ttyACM0 9600`
```
25
Read:
25, 1, 2, ...
```

This time you see the EEPROM dump contains the expected values.

6. Now flash eeprom_test26.bin `STM32_Programmer.sh -c port=USB1 -d eeprom_test26.bin 0x08000000 -g`
7. View serial again `screen /dev/ttyACM0 9600`
```
26
Read:
255, 255, 255...
```

See the EEPROM appears cleared. (This is the bug)
8. Reset and View serial again `screen /dev/ttyACM0 9600`
```
26
Read:
26, 1, 2, 3, 
```

Now it contains the expected values. This proves EEPROM is still "working".

**Expected behavior**
EEPROM should not be lost.

**Additional context**
Interestingly, if you roll back to eeprom_test25.bin the EEPROM is blank again (all 255's).

Even more interestingly if you then go back to eeprom_test26.bin the EEPROM is still at 26,1,2,3... as it was left.

Feels to me like the base EEPROM address define has changed. I've checked the changelogs and issues but can see anything that mentions this.

fpistm, 2023-11-10:
Hi @bobemoe 
I can't reproduce, it works as expected.
Anyway, my guess is that probably the second flash erase the flash, maybe STM32CubeProgrammer issue ?
As you stated, there is no change around the variant nor EEPROM.

bobemoe, 2023-11-10:
Hi @fpistm thanks for quick reply. 

Pretty sure neither of that is the issue, because the EEPROM is not erased on subsequent flashes as you can flash same version multiple times and its not cleared, or switch between the 2.5 and 2.6 version and the EEPROM for 2.6 is never cleared, but 2.5 always is, and then 2.6 re-appears after it seemed 2.5 cleared it! (See my notes in addition context)

Also cube programmer is only used in development with the binaries built locally.

In production we build in a Docker/Drone CI and flash OTA via ESP32 by following custom code following [AN3155](https://www.st.com/resource/en/application_note/cd00264342-usart-protocol-used-in-the-stm32-bootloader-stmicroelectronics.pdf) where we are also very careful not to erase the EEPROM. We are seeing the same issue when built and flashed like this too.

That said I just updated my cube programmer anyway 2.11 -> 2.14 and same issues still exists.

Maybe you could test with my binaries? [eeprom_test.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/13320803/eeprom_test.zip)

fpistm, 2023-11-10:
Got it. When adding STM32H5 support, I've update the EEPROMM library and this probably change the way STM32F4xx works. I have to deeper look.
https://github.com/stm32duino/Arduino_Core_STM32/commit/c71114382f7042e3397de0d66a7b8c30d595c87e#diff-ce86a4cc9d80ef1fb14663a45f154f7f4a4a948833a79e995636b62990260872

bobemoe, 2023-11-10:
Awesome! Let me know if you want me to test anything?

fpistm, 2023-11-10:
You can try to debug on your side ;)

bobemoe, 2023-11-10:
Well, I can confirm reverting the line in the commit you linked to above fixes the issue. 

Don't think I can debug much further though ;)

fpistm, 2023-11-10:
I  will debug next week. But it is possible the new version is the correct one.

bobemoe, 2023-11-10:
No worries, I can workaround now I know what it is.

I made a simpler test, just one byte, eliminated buffer functions, still get same results.

```c++
#include <EEPROM.h>
#include "version.h"

void setup() {
  Serial.begin(9600);
  while (!Serial) {;}

  Serial.print("Read:");
  Serial.print(EEPROM.read(0));
  Serial.println();

  Serial.print("Write:");
  Serial.println(VERSION);
  EEPROM.write(0,VERSION);
  Serial.println("Done.");
}

void loop() {
}
```

1. Flash eeprom_test25.bin
2. Run and confirm you see "Write:25"
3. Flash eeprom_tes26.bin
4. Run and confirm you must see "Read:25 \n Write:26" for working result.

Funnily enough I am actually debugging an internal issue of EEPROM resets (if byte 0 changes I apply migration or reset eeprom to default config) only intermittent (occasionally monthly 3/50 devices) could be hardware software whatever!?

Anyway we updated to see if it would help. Funny the update should introduce an identical symptom! I will pin back to 2.5.  I wonder if it did help. I wonder if they are related. Maybe your latest change solve my origin issue!

fpistm, 2023-11-11:
It is simple in fact.
I decrease the size of the FLASH used by EEPROM. In the 2.5 it was 16k and since 2.6 it is 8k.
That's why we get this issue.
https://github.com/stm32duino/Arduino_Core_STM32/commit/c71114382f7042e3397de0d66a7b8c30d595c87e#diff-ce86a4cc9d80ef1fb14663a45f154f7f4a4a948833a79e995636b62990260872L87

I will not revert to 16k. As the goal was to preserve maximum flash for code and avoid erase too much flash.
Anyway, you can redefine the size using by adding a build_opt.h file and add:
`-DFLASH_PAGE_SIZE=((uint32_t)(16*1024))`

Then it works as expected:
[image]

closed 2023-11-11 by fpistm

bobemoe, 2023-11-12:
Hi. Your resolution is great and thanks for providing a workaround.

Quick follow up question:
Where does the old data end up? The beginning of EEPROM must have changed address? Do you think it would be easy for me to make a function to copy the data to its new location before it gets destroyed by next page write? I'd prefer to migrate my code/data to the latest official version/config of things if possible.

fpistm, 2023-11-14:
> Hi. Your resolution is great and thanks for providing a workaround.
>
> Quick follow up question: Where does the old data end up? The beginning of EEPROM must have changed address? Do you think it [...]

With 2.5, I guess you can read your date, store it in another buffer, then refill at start address + 8kb. Flush in flash. Like this old data will be available within 2.6.

bobemoe, 2023-11-14:
> start address + 8kb.

Perfect, that is just what I needed to know. I will give that a try, Thanks.