stm32duino/Arduino_Core_STM32 #2180: declaring a class global variable causes `basepri` value to be set to a non-zero value

danyhm, 2023-11-12:
### Discussed in https://github.com/orgs/stm32duino/discussions/2178

<div type='discussions-op-text'>

<sup>Originally posted by **danyhm** November 10, 2023</sup>

**Describe the bug**
This bug only appears in STM32 boards and runs fine on AVR boards.
the problem is that when a simple global variable is added to the sketch, it seems to somehow messes with the PIN interrupt of
rc-switch library. also note that I'm including STM32Freertos.h. if you comment it out the bug is gone. I have no idea why this happens but in earlier versions of the board manager I never had this problem

**To Reproduce**
Steps to reproduce the behavior:
1- compile with line 17 commented.
2- upload to board and after pressing any remote button you should get some output
3- compile with line 17 UNcommented
4- upload to board and after pressing any remote button we don't get anything.

**Expected behavior**
adding the global variable should not have any effect.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [Windows 10 22H2 Enterprise]
 - Arduino IDE version: [2.2.1]
 - STM32 core version: [2.6.0]
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: [HID Bootloader]

**Board (please complete the following information):**
 - Name: [Bluepill]
 - Hardware Revision: [NA]
 - Extra hardware used if any: [433 rf receiver]

**Attached sketch**
[test_irq.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/13322811/test_irq.zip)
</div>

danyhm, 2023-11-12:
https://github.com/stm32duino/Arduino_Core_STM32/assets/21024667/9091ab9c-015e-4bf8-ac74-10fc4abe5531

fpistm, 2023-11-12:
https://github.com/orgs/stm32duino/discussions/2178#discussioncomment-7542007

fpistm, 2023-11-14:
This behavior is normal as you used STM32FreeRTOS.
When you include STM32FreeRTOS, it changes the way IRQ and some other ressources are managed.
As stated in the FAQ: https://www.freertos.org/FAQHelp.html
`If a FreeRTOS API function is called before the scheduler has been started then interrupts will deliberately be left disabled, and not re-enable again until the first task starts to execute. This is done to protect the system from crashes caused by interrupts attempting to use FreeRTOS API functions during system initialisation, before the scheduler has been started, and while the scheduler may be in an inconsistent state.`

So as alloc function is called it end up in the __sbrk:
https://github.com/stm32duino/STM32FreeRTOS/blob/f3abddcd420bd92a862b35877181392bd2647b2a/portable/MemMang/heap_useNewlib_ST.c#L159

So this is normal. You have to start the scheduler before init any thing relying on alloca api. FreeRTOS can also required some configuration depending of your use case but it is up to user to properly configure this.

closed 2023-11-14 by fpistm