stm32duino/Arduino_Core_STM32 #2224: FDCAN bus interrupts are not handled for the STM32G0B1 causing an infinite loop

rondlh, 2023-12-07:
With my limited understanding I see the following:

The FDCAN implementation on the STM32G0B1 uses interrupt "TIM16_FDCAN_IT0_IRQn", which calls "TIM16_IRQHandler(void)" (in HardwareTimer.cpp). 
In "timer.h" it says: #define TIM16_IRQHandler TIM16_FDCAN_IT0_IRQHandler.

"TIM16_IRQHandler" should analyze the interrupt source and forward to "HAL_FDCAN_IRQHandler" if the FDCAN bus is the interrupt source so the FDCAN callbacks can be used/called.

Currently TIM16_IRQHandler only processes interrupts triggered by Tim16, so the FDCAN bus callbacks cannot be used, and even worse, the interrupt flag is not cleared causing an infinite loop.

"TIM16_IRQHandler" in HardwareTimer.cpp
```
void TIM16_IRQHandler(void)
  {
    if (HardwareTimer_Handle[TIMER16_INDEX]) {
      HAL_TIM_IRQHandler(&HardwareTimer_Handle[TIMER16_INDEX]->handle);
    }
  }
```

fpistm, 2023-12-08:
Hi @rondlh 
right, as we have no official support for FDCAN it is not added.
Do not hesitate to provide a PR to implement this.

fpistm, 2024-03-11:
Hi  @rondlh,
I've made a PR to fix this issue, see #2301.
If you can test and give us a feedback, it would be very appreciated.