stm32duino/Arduino_Core_STM32 #2244: Linker error with version 2.7.1 for STM32 Nucleo F446RE

npalix, 2024-01-07:
The error  is 
```
arm-none-eabi-g++: warning: {compiler.optimization_flags}: linker input file unused because linking not done
arm-none-eabi-g++: error: {compiler.optimization_flags}: linker input file not found: No such file or directory

```

The issue comes from the platform file.
https://github.com/stm32duino/Arduino_Core_STM32/blob/2c45eba88a36cf95299ec44efe91ecb215fe06e4/platform.txt#L22

Setting the line to `compiler.optimization_flags={build.flags.optimize}` fixes the issue.

fpistm, 2024-01-08:
Hi @npalix
First happy new year.

There is no error on my side to build (and link) for the Nucleo F446RE. Moreover, the issue you point will simply impact all target which is not the case.
So you probably have an issue on your environment but you do not provide any information as requested by the issue template.

npalix, 2024-01-08:
**Describe the bug**
While compiling for any board the Blink example, I got the following error : 
```
arm-none-eabi-g++: warning: {compiler.optimization_flags}: linker input file unused because linking not done
arm-none-eabi-g++: error: {compiler.optimization_flags}: linker input file not found: No such file or directory

```

The issue seems to come from the platform file.
https://github.com/stm32duino/Arduino_Core_STM32/blob/2c45eba88a36cf95299ec44efe91ecb215fe06e4/platform.txt#L22

Setting the line to `compiler.optimization_flags={build.flags.optimize}` fixes the issue.

**To Reproduce**

1. Install following the instruction of https://github.com/stm32duino/Arduino_Core_STM32/wiki/Getting-Started
2. Load the Blink example
3. Compile

**Expected behavior**
Build binary for the current board

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Arduino IDE version: 1.8.19
 - STM32 core version: 2.7.1
 - Tools menu settings if not the default: default values
  
**Board (please complete the following information):**
 - Name: Nucleo F449RE

**Additional context**
I tried to remove and reinstall STM32Core.

fpistm, 2024-01-08:
As stated not reproduced:
[image]

Your logs seems weird because the arm none eabi binary has no full path.
Unfortunately, I can't help on this, maybe an Arduino IDE issue, you should try to remove all cache files and start from fresh install.

npalix, 2024-01-08:
The path is right with verbose mode (same place as in your screenshot). But it isn't prepended for warning/error messages (I guess g++ behavior). 

Can you provide information on how the flag is setup ?
It sounds odd for me to define it by itself and there is an experimental warning the line before. 
`compiler.optimization_flags={compiler.optimization_flags}`

My understanding is that the "{compiler.optimization_flags}" is not properly expanded and g++ interpretes it as a filename.

Moreover, the "optimize" value (-Ox) doesn't show up in the compilation command.

fpistm, 2024-01-09:
> Can you provide information on how the flag is setup ?
> It sounds odd for me to define it by itself and there is an experimental warning the line before.
> `compiler.optimization_flags={compiler.optimization_flags}`

@npalix You are right, the line is not correct. I've made a fix even if I can't reproduced (don't know why). This change make sense.
Thanks for your support.If you are able to test.
Note for the Arduino IDE 2.x, the impacted line is not used as only those one are used:
https://github.com/stm32duino/Arduino_Core_STM32/blob/2c45eba88a36cf95299ec44efe91ecb215fe06e4/platform.txt#L23C1-L24C41

depending of the `Sketch` -> `Optimize for Debugging` menu selection
[image]

PR #2247 (2024-01-09): fix(platform): wrong optimization_flags for Arduino 1.8.x
Introduced by #2184.
Fixes #2244.

Even if not reproduced, the recipe is not coherent. Only impact Arduino 1.8.x.
commit 6e619060 (2024-01-09): fix(platform): wrong optimization_flags for Arduino 1.8.x

Introduced by #2184.
Fixes #2244.
closed 2024-01-09 by fpistm