stm32duino/Arduino_Core_STM32 #2277: Unable to use debug function in Arduino IDE 2.3.0

China-Mr-Hou, 2024-02-15:
**Please, before reporting any issue**
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](https://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).
- **Avoid** to submit a GitHub issue for project troubleshooting.

Any questions/feedback/suggestions should be discussed on the [stm32duino forum](https://stm32duino.com):
  * questions on the [STM32 Core](https://www.stm32duino.com/viewforum.php?f=35)
  * bugs/enhancements on the [STM core: Bugs and enhancements](https://www.stm32duino.com/viewforum.php?f=38)

&#x26A0; When reporting any issue, please try to provide all relevant information to help on its resolution.

**Describe the bug**
Recently I'm using Arduino IDE version 2.2.1, its debug function worked well on my H743VIT6, however after I upgrade to 2.3.0, it says I should select a programmer and doesn't show any more info. The “Tool" menu above doesn't have a programmer selection either. I tried copying the programmer.txt file included from Arduino GIGA R1 package from Arduino Official since GIGA board also uses STM32H7 as the core chip and it didn't change anything.

**To Reproduce**

Steps to reproduce the behavior:
1. Install Arduino IDE 2.3.0
2. Click on the Debug Tab one the left side
3. See error: No programmer selected for XXX

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11 build 22621
 - Arduino IDE version: 2.3.0
 - STM32 core version: 2.7.1
 - Tools menu settings if not the default: Nothing changed.
 - Upload method: SWD

**Board (please complete the following information):**
 - Name: Generic H743VITx
 - Hardware Revision: None
 - Extra hardware used if any: None

**Additional context**
I have standalone OpenOCD and JLink software installed.

fpistm, 2024-02-15:
Hi @China-Mr-Hou 
I saw this issue. It comes from changes from arduino-cli.
I've already started to work on this but need more time to properly handle that and I think I found some issues with the Arduino IDE 2

China-Mr-Hou, 2024-02-15:
I wrote this comment after I have upgraded my IDE to 2.3.1 and found Arduino Team didn't fix this problem (since issue#2354 is intended). In my opinion, they may be thinking choosing a programmer instead of writing debug configuration json file is simpler. Anyway, I still appreciate the Arduino developing platform itself and every third party contributions for making hardware coding easy for everyone.

fpistm, 2024-02-20:
Hi @China-Mr-Hou 
the issue will be solved within next Arduino IDE released (refers to the https://github.com/arduino/arduino-ide/issues/2368).
Anyway, I will reviewed all the debug configuration to enhance user experience.

closed 2024-02-20 by fpistm