stm32duino/Arduino_Core_STM32 #2281: Can't re-define ISR handler for interrupts/exceptions defined as weak in startup_stm32f103x6.s

mha1, 2024-02-20:
**Describe the bug**
I am trying to bare-metal re-define the ISR for TIM1 locally by 

```C
extern "C" {
  void TIM1_UP_IRQHandler(void) {

    if (TIM1->SR & TIM_SR_UIF) {
      TIM1->SR &= ~(TIM_SR_UIF);
      // do something
    }
  }
}
```

TIM1 interrupts still end up being serviced by `TIM1_IRQHandler(void)` defined in `HardwareTimer.cpp` . The linker chooses `HardwareTimer.cpp`'s `TIM1_IRQHandler(void)` over the locally defined ISR despite weak definitions in `startup_stm32f103x6.s`
```
  .weak TIM1_UP_IRQHandler
  .thumb_set TIM1_UP_IRQHandler,Default_Handler
```

Adding __weak to the definition of `HardwareTimer.cpp`'s ISR solves the problem:
```C
...
#if defined(TIM1_BASE)
  /**
    * @brief  TIM1 IRQHandler common with TIM10 and TIM16 on some STM32F1xx
    * @param  None
    * @retval None
    */
  __weak void TIM1_IRQHandler(void)
  {
    if (HardwareTimer_Handle[TIMER1_INDEX]) {
      HAL_TIM_IRQHandler(&HardwareTimer_Handle[TIMER1_INDEX]->handle);
    }
...
```

I think ISR/exception handler implementations for all interrupts/exceptions defined as weak in startup.s files should be defined __weak too.

**To Reproduce**
- try to bare-metal re-define any STM32 exception/interrupt handler locally

**Expected behavior**
- being able to bare-metal re-define any STM32 exception/interrupt handler locally

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino IDE version: 
   VSCode & PlatformIO with
   framework-arduinoststm32 @ 4.10900.200819 (1.9.0)
 - Upload method: JLink

**Board:**
 - Name: generic STM32F103 (bluepill) but this is a general problem for all STM32 variants

fpistm, 2024-02-20:
As you stated the weak function is defined in the startup. The one defined in HardwareTimer.cpp well redefined it and it is expected. There is no issue and no adding weak to the strong one is not correct because you can't tell which one will be used.

If you want to do bare metal you can simply remove the Arduino API part using the TIM IRQ:
https://github.com/stm32duino/Arduino_Core_STM32/wiki/HAL-configuration#hal-module-only

So in your case `HAL_TIM_MODULE_ONLY`.

closed 2024-02-20 by fpistm

mha1, 2024-02-20:
Thanks, works fine

fpistm, 2024-02-20:
> Thanks, works fine

Welcome 😉