stm32duino/Arduino_Core_STM32 #2308: Servo.attach() call to Servo::write() before min/max are configured leads to issue when writting angles in degrees

jan019, 2024-03-19:
**Describe the bug**
The Servo library calls the `write() `function within the `attach()` function https://github.com/stm32duino/Arduino_Core_STM32/blob/641f3d8883e9a7220f65dded355dc32e218a990d/libraries/Servo/src/stm32/Servo.cpp#L124 before configuring ` this->min` and `this->max` values https://github.com/stm32duino/Arduino_Core_STM32/blob/641f3d8883e9a7220f65dded355dc32e218a990d/libraries/Servo/src/stm32/Servo.cpp#L126-L127, which however need to be configured for `write()` function to work correctly when the servo position is set in degrees. The function `write() `uses the `this->min` and `this->max` values to map the passed value in degrees between min and max https://github.com/stm32duino/Arduino_Core_STM32/blob/641f3d8883e9a7220f65dded355dc32e218a990d/libraries/Servo/src/stm32/Servo.cpp#L156/

**To Reproduce**
```int pin = 1;
int min = 40;
int max = 70;
int value = 40;

// on first call to attach, the write is called with a different value in this->min and this->max than on following calls
Servo.attach(pin, min, max, value)
```

 - STM32 core version: 2.7.1

The write() function should either not be called in the attach function at all, or it must be called after `this->min` and `this->max` are configured.

fpistm, 2024-03-20:
Hi @jan019 
Thanks for the detailed report.
It seems logical. Could you submit a PR ? I guess configuring the values before the write call instead of after is enough.

jan019, 2024-03-21:
Hi @fpistm, thanks for checking this out. I will submit a PR later today with configuring the values before the write call instead of after (I have tested this and it works well).

PR #2318 (2024-03-25): Fix issue with writing inital servo value before min/max is configured
**Summary**

The call to function Servo::write() in Servo::attach() must be done after Servo::min and Servo::max are configured - Servo::write() uses the min/max for mapping the passed value if the value is passed in degrees. Therefore if inital value in attach is set in degress and min/max is not configured when call to Servo::write() is done, the mapping will not be correct.

Fixes: #2308
commit f67f03b5 (2024-03-25): Fix issue with writing inital servo value before min/max is configured

The call to function Servo::write() in Servo::attach() must be done
after Servo::min and Servo::max are configured - Servo::write() uses the
min/max for mapping the passed value if the value is passed in degrees.
Therefore if inital value in attach is set in degress and min/max
is not configured when call to Servo::write() is done, the mapping will
not be correct.

Resolves: #2308
closed 2024-03-25 by fpistm