stm32duino/Arduino_Core_STM32 #2312: While debugging, cannot access registers

MasteringTheMess, 2024-03-20:
When debugging using Arduino IDE 2.3.2 and STM32core 2.7.1 (fresh install of both), I have found no way to get information of the hardware registers.

Simply adding RCC or TIM2 to the watch list does not work

Next to the variabel name appears the text:

> <-var-create: unable to create variable object (from var-create  watch_981046eac58f803c608b22be9fee90d9382b4419ffc812193eac6fa47970566f @ "tempreg")>

and, when stopping the debugger, in the debug console appears this error message:

> Thread Warning: watch: eval. expression 'RCC' with no thread context. Using default
> Thread Warning: watch: eval. expression 'RCC' with no thread context. Using default
> Thread Warning: watch: eval. expression 'TIM2' with no thread context. Using default
> Thread Warning: watch: eval. expression 'TIM2' with no thread context. Using default

The variables are there and can be used, I do configure the registers

full code (ontop of blink-example)
```
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(921600);

	uint32_t tempreg = 0;
	pinMode(PA5, INPUT_PULLUP);
	RCC->APB1ENR1 |= RCC_APB1ENR1_TIM2EN;
	LL_TIM_InitTypeDef TIM_InitStruct = {0};

	TIM_InitStruct.Prescaler = 0;
	TIM_InitStruct.CounterMode = LL_TIM_COUNTERMODE_UP;
	TIM_InitStruct.Autoreload = 0xFFFF;
	TIM_InitStruct.ClockDivision = LL_TIM_CLOCKDIVISION_DIV1;
	TIM_InitStruct.RepetitionCounter = 0;
	LL_TIM_Init(TIM2, &TIM_InitStruct);
	LL_TIM_DisableARRPreload(TIM2);
	LL_TIM_SetTriggerOutput(TIM2, LL_TIM_TRGO_RESET);
//	LL_TIM_DisableMasterSlaveMode(TIM2);

	LL_TIM_IC_SetActiveInput(TIM2, LL_TIM_CHANNEL_CH1, LL_TIM_ACTIVEINPUT_DIRECTTI);
	LL_TIM_IC_SetPrescaler(TIM2, LL_TIM_CHANNEL_CH1, LL_TIM_ICPSC_DIV1);
	LL_TIM_IC_SetFilter(TIM2, LL_TIM_CHANNEL_CH1, LL_TIM_IC_FILTER_FDIV1_N8);
	LL_TIM_IC_SetPolarity(TIM2, LL_TIM_CHANNEL_CH1, LL_TIM_IC_POLARITY_RISING);

	LL_TIM_IC_SetActiveInput(TIM2, LL_TIM_CHANNEL_CH2, LL_TIM_ACTIVEINPUT_DIRECTTI);
	LL_TIM_IC_SetPrescaler(TIM2, LL_TIM_CHANNEL_CH2, LL_TIM_ICPSC_DIV1);
	LL_TIM_IC_SetFilter(TIM2, LL_TIM_CHANNEL_CH2, LL_TIM_IC_FILTER_FDIV1_N8);
	LL_TIM_IC_SetPolarity(TIM2, LL_TIM_CHANNEL_CH2, LL_TIM_IC_POLARITY_FALLING);

	LL_TIM_SetTriggerInput(TIM2, LL_TIM_TS_TI1FP1);
	LL_TIM_SetSlaveMode(TIM2, LL_TIM_SLAVEMODE_COMBINED_RESETTRIGGER);

	LL_TIM_EnableCounter(TIM2);

	Serial.print("TIM2->CCMR1: ");
	Serial.println(TIM2->CCMR1);
	tempreg = TIM2->CCMR1;
  Serial.println(tempreg);

}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}

```

fpistm, 2024-03-20:
You have to add svd file using debug_custon.json file. See the forum to know how to write json file.

closed 2024-03-20 by fpistm

MasteringTheMess, 2024-03-20:
I created a custom_debug.json with contents
```
{
    "svdFile": "C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeProgrammer\SVD\STM32L4R5.svd"
}
```
in the sketch folder, but that did not work.

Then I changed the json to
```
{
    "svdFile": "STM32L4R5.svd"
}
```
 and copied STM32L4R5.svd into the sketch folder, but it still does not work.

MasteringTheMess, 2024-03-20:
I also tried using different slashes for the path, no luck either ...

fpistm, 2024-03-20:
In fact since IDE 2.3.x, the debug has been reworked by Arduino.
I'm currently working to support the new way to support debug svd files.

fpistm, 2024-03-20:
For ref: https://github.com/arduino/arduino-ide/issues/1920

MasteringTheMess, 2024-03-21:
I downgraded to 2.0.3 but I still don't get access to peripheral registers
So clearly, I am still doing something wrong :/

MasteringTheMess, 2024-03-21:
Oh my...
Just realized that you don't have to manualy create a watch expression, but that there is a complete peripheral register viewer...

...and for those who use sloeber
you can specify a svd-file to use in the debug launch configuration:
[image]

fpistm, 2024-03-21:
Here a way to get the svd properly load within Arduino IDE 2.3.2:
In the boards.txt add this line to the board entry you want:

Example for Nucleo L496ZG-P
`Nucleo_144.menu.pnum.NUCLEO_L496ZG-P.debug.svd_file=C:\\ST\STM32CubeCLT_1.15.0\\STMicroelectronics_CMSIS_SVD\\STM32L4x6.svd
`

[image]