stm32duino/Arduino_Core_STM32 #2313: on debug, variable gets optimized out

MasteringTheMess, 2024-03-20:
While debugging in Arduino IDE 2, a local variable gets optimized out even if it is used
optimization is set to -g or -0, symbols and core logs enabled

[image]

Arduino IDE 2.3.2
STM32 core 2.7.1
Nucleo-L4R5ZI-P

full source code:
```
void setup() {
	uint32_t tempreg = 0;
  Serial.begin(921600);

	pinMode(PA5, INPUT_PULLUP);
	RCC->APB1ENR1 |= RCC_APB1ENR1_TIM2EN;
	LL_TIM_InitTypeDef TIM_InitStruct = {0};

	TIM_InitStruct.Prescaler = 0;
	TIM_InitStruct.CounterMode = LL_TIM_COUNTERMODE_UP;
	TIM_InitStruct.Autoreload = 0xFFFF;
	TIM_InitStruct.ClockDivision = LL_TIM_CLOCKDIVISION_DIV1;
	TIM_InitStruct.RepetitionCounter = 0;
	LL_TIM_Init(TIM2, &TIM_InitStruct);
	LL_TIM_DisableARRPreload(TIM2);
	LL_TIM_SetTriggerOutput(TIM2, LL_TIM_TRGO_RESET);
	tempreg = TIM2->CCMR1;

	LL_TIM_IC_SetActiveInput(TIM2, LL_TIM_CHANNEL_CH1, LL_TIM_ACTIVEINPUT_DIRECTTI);
	LL_TIM_IC_SetPrescaler(TIM2, LL_TIM_CHANNEL_CH1, LL_TIM_ICPSC_DIV1);
	LL_TIM_IC_SetFilter(TIM2, LL_TIM_CHANNEL_CH1, LL_TIM_IC_FILTER_FDIV1_N8);
	LL_TIM_IC_SetPolarity(TIM2, LL_TIM_CHANNEL_CH1, LL_TIM_IC_POLARITY_RISING);
	tempreg = TIM2->CCMR1;

	LL_TIM_IC_SetActiveInput(TIM2, LL_TIM_CHANNEL_CH2, LL_TIM_ACTIVEINPUT_DIRECTTI);
	LL_TIM_IC_SetPrescaler(TIM2, LL_TIM_CHANNEL_CH2, LL_TIM_ICPSC_DIV1);
	LL_TIM_IC_SetFilter(TIM2, LL_TIM_CHANNEL_CH2, LL_TIM_IC_FILTER_FDIV1_N8);
	LL_TIM_IC_SetPolarity(TIM2, LL_TIM_CHANNEL_CH2, LL_TIM_IC_POLARITY_FALLING);
	tempreg = TIM2->CCMR1;

	LL_TIM_SetTriggerInput(TIM2, LL_TIM_TS_TI1FP1);
	LL_TIM_SetSlaveMode(TIM2, LL_TIM_SLAVEMODE_COMBINED_RESETTRIGGER);

	LL_TIM_EnableCounter(TIM2);

	Serial.print("TIM2->CCMR1: ");
	Serial.println(TIM2->CCMR1);
	tempreg = TIM2->CCMR1;
  Serial.println(tempreg);

}

// the loop function runs over and over again forever
void loop() {
}
```

fpistm, 2024-03-20:
Use optimize for debug.

closed 2024-03-20 by fpistm

MasteringTheMess, 2024-03-20:
I do !

as stated above 

[image]

[image]

fpistm, 2024-03-20:
In that case nothing I can do. At the step you reach variable is probably not useful so optimised out