stm32duino/Arduino_Core_STM32 #2324: Nucleo-F439ZI support

pmantoine, 2024-04-01:
STM Nucleo-F439ZI is identical to the F429ZI with the only difference being that it has crypto hardware in the CPU itself.

As such... it just needs the Nucleo-F429ZI variant and board files copied and appropriately updated with 439 instead of 429 inserted in all the relevant places.

I have done this manually myself... happy to contribute the files if desired.

**Note that user can add a STM32 based board following this [wiki](https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-(board))**

Any contribution is welcome, so do not hesitate to submit a PR.

All new variant request will be moved to [New Variant] request list #722 and the related issue will be closed.

fpistm, 2024-04-01:
Hi @pmantoine 
Feel free to submit a PR.

pmantoine, 2024-04-01:
Will do.

fpistm, 2024-04-20:
Added to #722.

closed 2024-04-20 by fpistm

pmantoine, 2024-04-21:
Issued PR as requested, apologies for the delay.

PR #2347 (2024-04-24): Initial support for Nucleo-F439ZI

* [x] Variant files for Nucleo-F439ZI device support

**Closing issues**

Fixes #2324