stm32duino/Arduino_Core_STM32 #2338: STM32F030CCT6 PB15 GPIO Interrupt

xiezhoubin, 2024-04-15:
Arduino developed stm32f030cct6, which cannot use PB15 as an external interrupt pin. Once attachInterrupt (digitally PinToInterrupt (PB15), callback, CHANGE) is used; This line of code will cause the program to get stuck here

fpistm, 2024-04-15:
Hi @xiezhoubin 
Could you be more precise?
Sketch? Board used? Core version?....

xiezhoubin, 2024-04-15:
``> Hi @xiezhoubin Could you be more precise? Sketch? Board used? Core version?....

chip:stm32f030cct6

Core version:2.7.1

code:
```
#include <Arduino.h>

HardwareSerial mySerial(PA10, PA9);

volatile uint32_t count = 0;
volatile bool flag = false;

void callback() {
  if (flag) {
    count++;
  }
}

void setup() {
  delay(1000);

  mySerial.begin(115200);
  mySerial.setTimeout(5);
  mySerial.println("\n start...");

  pinMode(groundSenseDetectionPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(PB15), callback, CHANGE); 
  mySerial.println("\n ok");
}

void loop() {
  mySerial.println("count:" + String(count));
  delay(100);
}
```

closed 2024-04-15 by fpistm