stm32duino/Arduino_Core_STM32 #2369: `HardwareSerial::flush()` returns immediately if called as `Stream*->flush()` interface

mlaga97, 2024-05-14:
**Please, before reporting any issue**
- Make sure you are using the latest Arduino_Core_STM32 version.
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](https://stm32duino.com)
- Submit a GitHub issue only for reporting a problem related to the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).
- **Avoid** to submit a GitHub issue for project troubleshooting.

Any questions/feedback/suggestions should be discussed on the [stm32duino forum](https://stm32duino.com):
  * questions on the [STM32 Core](https://www.stm32duino.com/viewforum.php?f=35)
  * bugs/enhancements on the [STM core: Bugs and enhancements](https://www.stm32duino.com/viewforum.php?f=38)

&#x26A0; When reporting any issue, please try to provide all relevant information to help on its resolution.

**Describe the bug**
`HardwareSerial::flush()` returns early if called via Stream* interface

**To Reproduce**
```
const int PIN_EN = PA1;
const int PIN_TX = PA2;
const int PIN_RX = PA3;

HardwareSerial Serial1(PIN_RX, PIN_TX);
Stream* streamTest = &Serial1;

void setup() {
  Serial1.begin(115200);
  pinMode(PIN_EN, OUTPUT);
}

void loop() {
  // Calling directly
  Serial1.print("aaaaaa");

  digitalWrite(PIN_EN, HIGH);
  Serial1.flush();
  digitalWrite(PIN_EN, LOW); // Goes low only after data has been fully transmitted

  delay(1);

  // Calling via generic stream interface
  streamTest->print("bbbbbb");

  digitalWrite(PIN_EN, HIGH);
  streamTest->flush();
  digitalWrite(PIN_EN, LOW); // Goes low immediately

  delay(1);
}
```

**Expected behavior**
In the above code, `streamTest->flush()` should return only once the data has been fully flushed.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04
 - Arduino IDE version: [e.g. 1.8.8]
 - STM32 core version: 2.7.1, issue appears to be a result of #2124
 - Tools menu settings if not the default: default
 - Upload method: ST-LINKv2

**Board (please complete the following information):**
 - Name: Generic STM32G030
 - Hardware Revision: N/A
 - Extra hardware used if any: N/A

**Additional context**
This appears to be a side effect of #2124, as reverting that commit results in the issue going away.

It seems as though the optional parameter is preventing `HardwareSerial::flush(uint32_t timeout = 0)` from properly overloading `Stream::flush(void)`.

Therefore, another method of resolving is by modifying HardwareSerial.h from
```
virtual void flush(uint32_t timeout = 0);
``` 
to
```
virtual void flush()
{
  flush(0);
}
virtual void flush(uint32_t timeout);
```

mlaga97, 2024-05-14:
Need to search better. #2254

closed 2024-05-14 by mlaga97

fpistm, 2024-05-14:
Hi @mlaga97 
No worry, fix will be available within next release in the coming weeks.