stm32duino/Arduino_Core_STM32 #2400: Nucleo 32 G431KB: SPI not working as expected

acunipisa, 2024-06-14:
**Describe the bug**
Using a Nucleo 32 stm32g431kb for SPI communication. After removing SB4 jumper (to disconnect PB3, or D13, from SWO of the on board stlink), SPI is not working as expected using stmduino core.
No signal on SCK (D13, or PB3) can be observed using an o-scope. The same is true for MOSI (D11, or PB5).

**To Reproduce**
Nucleo32 gk431kb with SB4 jumper removed. Plug USB, and upload the code using Arduino IDE 2.3.2. 
```
#include <SPI.h>

void setup() {

  SPI.begin(); 
  SPI.beginTransaction(SPISettings(10000000, MSBFIRST, SPI_MODE0)); 
  SPI.transfer(0x52); 

}

void loop() {

  delay(10);
  SPI.beginTransaction(SPISettings(10000000, MSBFIRST, SPI_MODE0));
  SPI.transfer(0x52); 

}
```

**Expected behavior**
Normally, I would expect the SPI lines to show signals according to the communication protocol.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.6.0
 - Tools menu settings if not the default: default
 - Upload method: Mass Storage

**Board (please complete the following information):**
 - Name: Nucleo 32 G431KB

fpistm, 2024-06-14:
Hi @acunipisa 
You didn't change the pins used for SPI instance.
https://github.com/stm32duino/Arduino_Core_STM32/blob/7efecbf271696d65da88282c8187a23609afbf66/variants/STM32G4xx/G431K(6-8-B)(T-U)_G441KB(T-U)/variant_NUCLEO_G431KB.h#L106

See the wiki to know how do this.

acunipisa, 2024-06-14:
I appreciate your support.

Indeed, I changed the pins using the following code, but still not signals on the lines. I am wondering if it has something to do with the JTAG interface.

```
#include "SPI.h"

void setup() {

  SPI.setMISO(PB4);
  SPI.setMOSI(PB5);
  SPI.setSCLK(PB3);

  SPI.begin(1);
}

void loop() {

  delay(10);
  SPI.transfer(0x52);

}
```

fpistm, 2024-06-14:
Why this:
`SPI.begin(1);`
it should be no argument. Which core version you used ?

acunipisa, 2024-06-14:
Using core 2.6.0. I may try to use the new one. 
Without the "1" the code just freezes if I use the same code.

fpistm, 2024-06-14:
Even with 2.6.0 you should not use argument for begin.

acunipisa, 2024-06-14:
Installed core 2.7.1. 

Removed the "1". If I do not modify the pins, I can observe the signals on PA7 and PA5, as expected.

However, when I change the pins, the code gets stuck when I set PB3 as SPI CLK. If I do not, MOSI is correctly observed on PB5.

```
#include "SPI.h"

void setup() {

  SPI.setMISO(PB4);
  SPI.setMOSI(PB5);
  //SPI.setSCLK(PB3); // uncomment and the code gets stuck

  SPI.begin();
}

void loop() {

  delay(10);
  SPI.transfer(0x52);

}
```

acunipisa, 2024-06-14:
> Installed core 2.7.1.
>
> Removed the "1". If I do not modify the pins, I can observe the signals on PA7 and PA5, as expected.
>
> However, when I change the pins, the code gets stuck when I set PB3 as [...]

By the way, the SCLK cannot be routed to PB3, and I cannot understand why. From the stm32g431kb datasheet it should be possible (indeed the manual of the nucleo32 board and the layout suggest that as well). I will keep investigating.

fpistm, 2024-06-14:
I've made a test with a Nucleo G431RB as I do not have the KB (but similar) and it works:
[image]

Will try to find a KB but my gess is you damage your board when removing the SB.

acunipisa, 2024-06-14:
First, let me thank you for taking the time to test the code on another board.

Your tests inspired me to try programming the Nucleo32 as a generic G431KB board variant, and it worked like a charm. However, when I tried again as a Nucleo32, it failed.

Therefore, the issue only occurs when the Nucleo32 variant is chosen.

I can live with this workaround, but wanted to let you know.

fpistm, 2024-06-14:
Really strange. Will check that.

fpistm, 2024-06-17:
OK. Found the issue, it misses some `PinName `in the `digitalPin `array.
See #2043.

acunipisa, 2024-06-17:
> OK. Found the issue, it misses some `PinName `in the `digitalPin `array. See #2043.

Great. Thank you for your work. Bests

fpistm, 2024-06-17:
If you can test to validate this is the only issue 😉 
I'm confident this will fix your issue.

acunipisa, 2024-06-17:
> If you can test to validate this is the only issue 😉 I'm confident this will fix your issue.

Tested, It works! Thank you!

commit ced1323c (2024-06-17): fix(nucleo_g431k): add missing PinName in digitalPin array

Fixes #2400.
closed 2024-06-17 by fpistm