stm32duino/Arduino_Core_STM32 #2402: Compiler warnings about -Wregister in STM32C0x

maxgerhardt, 2024-06-15:
**Describe the bug**
Compiling for STM32C0x produces tons of register warnings in even the latest git commit version. They are all about the same  
```
warning: ISO C++17 does not allow 'register' storage class specifier 
```
type.

```
arm-none-eabi-g++ -o .pio\build\genericSTM32C011F6\SrcWrapper\src\stm32\interrupt.cpp.o -c -std=gnu++17 -fno-threadsafe-statics -fno-rtti -fno-exceptions -fno-use-cxa-atexit -mcpu=cortex-m0plus -mthumb -Os -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DPLATFORMIO=60115 -DSTM32C0 -DSTM32C011xx -DARDUINO_GENERIC_C011F6PX -DSTM32C0xx -DARDUINO=10808 -DARDUINO_ARCH_STM32 -DNDEBUG -DARDUINO_GENERIC_C011F6PX -DBOARD_NAME=\"GENERIC_C011F6PX\" -DHAL_UART_MODULE_ENABLED -DUSE_HAL_DRIVER -DUSE_FULL_LL_DRIVER -DVARIANT_H=\"variant_generic.h\" -DVECT_TAB_OFFSET=0x0 [...]
In file included from C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_hal_adc.h:31,
                 from C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\STM32C0xx/stm32c0xx_hal_conf_default.h:223,
                 from C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\STM32C0xx/stm32c0xx_hal_conf.h:13,
                 from C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_hal.h:30,
                 from C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32C0xx\Include/stm32c0xx.h:197,
                 from C:\Users\Max\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/stm32_def.h:24,
                 from C:\Users\Max\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/interrupt.h:43,
                 from C:\Users\Max\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32\interrupt.cpp:38:
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h: In function 'void LL_ADC_SetAnalogWDMonitChannels(ADC_TypeDef*, uint32_t, uint32_t)':
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h:3677:27: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
 3677 |   register __IO uint32_t *preg;
      |                           ^~~~
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h: In function 'uint32_t LL_ADC_GetAnalogWDMonitChannels(ADC_TypeDef*, uint32_t)':
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h:3781:27: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
 3781 |   register __IO uint32_t *preg = __ADC_PTR_REG_OFFSET(ADCx->CFGR1, ((AWDy & ADC_AWD_CRX_REGOFFSET_MASK) >> ADC_AWD_CRX_REGOFFSET_POS)
      |                           ^~~~
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h:3784:21: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
 3784 |   register uint32_t AnalogWDMonitChannels = (READ_BIT(*preg, AWDy) & AWDy & ADC_AWD_CR_ALL_CHANNEL_MASK);
      |                     ^~~~~~~~~~~~~~~~~~~~~
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h: In function 'void LL_ADC_ConfigAnalogWDThresholds(ADC_TypeDef*, uint32_t, uint32_t, uint32_t)':
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h:3903:27: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
 3903 |   register __IO uint32_t *preg = __ADC_PTR_REG_OFFSET(ADCx->AWD1TR, (((AWDy & ADC_AWD_TRX_REGOFFSET_MASK)) >> (ADC_AWD_TRX_REGOFFSET_BITOFFSET_POS)) + ((ADC_AWD_CR3_REGOFFSET & AWDy) >> (ADC_AWD_CRX_REGOFFSET_BITOFFSET_POS + 1UL)));
      |                           ^~~~
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h: In function 'void LL_ADC_SetAnalogWDThresholds(ADC_TypeDef*, uint32_t, uint32_t, uint32_t)':
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h:3980:27: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
 3980 |   register __IO uint32_t *preg = __ADC_PTR_REG_OFFSET(ADCx->AWD1TR, (((AWDy & ADC_AWD_TRX_REGOFFSET_MASK)) >> (ADC_AWD_TRX_REGOFFSET_BITOFFSET_POS)) + ((ADC_AWD_CR3_REGOFFSET & AWDy) >> (ADC_AWD_CRX_REGOFFSET_BITOFFSET_POS + 1UL)));
      |                           ^~~~
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h: In function 'uint32_t LL_ADC_GetAnalogWDThresholds(ADC_TypeDef*, uint32_t, uint32_t)':
C:\Users\Max\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32C0xx_HAL_Driver\Inc/stm32c0xx_ll_adc.h:4022:33: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
 4022 |   register const __IO uint32_t *preg = __ADC_PTR_REG_OFFSET(ADCx->AWD1TR, (((AWDy & ADC_AWD_TRX_REGOFFSET_MASK)) >> (ADC_AWD_TRX_REGOFFSET_BITOFFSET_POS)) + ((ADC_AWD_CR3_REGOFFSET & AWDy) >> (ADC_AWD_CRX_REGOFFSET_BITOFFSET_POS + 1UL)));
      |                                 ^~~~
```

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
1. Download https://github.com/maxgerhardt/pio-stm32c0
2. Open it in VSCode with the PlatformIO extension installed
3. Delete line 15 (`extra_scripts = surpress_register_warnings.py`)
4. [Project Task](https://docs.platformio.org/en/latest/integration/ide/vscode.html#project-tasks) -> Build
5. Observe errors

**Expected behavior**
Clean build, no warnings.

**Screenshots**
-/-

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino IDE version: Not applicable, PlatformIO core  6.1.15
 - STM32 core version: Git hash 361a7fdb67e2a7104e99b4f42a802469eef8b129
 - Tools menu settings if not the default: Default
 - Upload method: SWD

**Board (please complete the following information):**
 - Name: Generic C011F6Px
 - Hardware Revision: None
 - Extra hardware used if any: None 

**Additional context**
This was already raised in https://github.com/stm32duino/Arduino_Core_STM32/issues/2250#issue-2077815696 but closed and not fixed.

fpistm, 2024-06-15:
Right this is linked to HAL/LL  from the cube. This should be reported to STMICROELECTRONICS GitHub.  Here we only use it no change. This will bre fixed within future update when fixed in the cube.

closed 2024-06-15 by fpistm

fpistm, 2024-06-17:
You can file issue here to track progress:
https://github.com/STMicroelectronics/STM32CubeC0/issues

And here a list of issue already opened for other series:
https://github.com/search?q=org%3ASTMicroelectronics+%27register%27+storage+class+specifier+%5B-Wregister%5D&type=issues

maxgerhardt, 2024-06-17:
The response I got from STMicroelectronics "support" is very "don't bother us" like.
```
Case#: 00208029
Subject: Using STM32C0xx HAL with C++ generates tons of warnings
Description:
Dear Mr. Max,

thank you for contacting ST support. register keyword in C "hints" compiler to store variable in core register, if possible. Nevertheless, since ISO C++17 is usage of this keyword not supported and should not be used (however in C still OK). A simple solution may be removing the register keyword from the declarations.

With regards,
ST mcu support
```

fpistm, 2024-06-17:
Unfortunately can do nothing here. Anyway I guess it will  be removed as it is done for other series.