stm32duino/Arduino_Core_STM32 #2412: ELV-BM-TRX1 was not affected by the rework for the STM32Programmer

fabry1992, 2024-06-28:
**Describe the bug**
Programming the ELV-BM-TRX1 device results in following error message:
`Selected interface: 1
Protocol unknown!`

**To Reproduce**
1. Use the compile function
2. See message 

**Expected behavior**
The compiled code gets Downloaded and verified onto the device

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.8.0
 - Upload method: Serial

**Board (please complete the following information):**
 - Name: ELV-BM-TRX1
 - Hardware Revision: Rev A

**Suggested Fix**
Since all other board.txt entries were reworked, in the same way, I would suggest to change the following lines:
`12946 : ELV_Modular_System.menu.upload_method.swdMethod.upload.protocol=0`
`12950 : ELV_Modular_System.menu.upload_method.serialMethod.upload.protocol=1`
`12951 : ELV_Modular_System.menu.upload_method.serialMethod.upload.options={serial.port.file}`

to 

`12946 : ELV_Modular_System.menu.upload_method.swdMethod.upload.protocol=swd`
`12950 : ELV_Modular_System.menu.upload_method.serialMethod.upload.protocol=serial`
`12951 : ELV_Modular_System.menu.upload_method.serialMethod.upload.options=-c {serial.port.file}`

fpistm, 2024-06-28:
Hi @fabry1992 
Right, 2 PR were made in // and forgot this new entry for the STM32CubeProgrammer rework. Plase feel free to provide a PR.
As a workaround, the boards.local.txt can be used to redefine the correct options.

fpistm, 2024-07-01:
Workaround:

1. Create a [`boards.local.txt`](https://arduino.github.io/arduino-cli/0.27/platform-specification/#boardslocaltxt) file.
2. Add those lines inside:
```keyvalues
ELV_Modular_System.menu.upload_method.swdMethod.upload.protocol=swd
ELV_Modular_System.menu.upload_method.serialMethod.upload.protocol=serial
ELV_Modular_System.menu.upload_method.serialMethod.upload.options=-c {serial.port.file}
```

PR #2414 (2024-07-01): fix: ELV_Modular_System upload options
Fixes #2412
commit 298889fa (2024-07-01): fix: ELV_Modular_System upload options

Fixes #2412
commit e194ab86 (2024-07-01): fix: ELV_Modular_System upload options

Fixes #2412
closed 2024-07-01 by fpistm