stm32duino/Arduino_Core_STM32 #2413: I2C, UART and Deepsleep on STM32wle5jc(Wio-E5 mini) work only in Debug

miroslavpetrov, 2024-07-01:
**Describe the bug**
I have based my sketch on the LowPower example from the LoRaWAN library. With the bare example, the timer wake from sleep was not working and the board was sleeping for only 600ms. Reading other user's issues I have added a call to rtc.begin(true) after modem.begin() in setup and this fixed the issue. With the bare example, Serial.print() also doesn't work after the first wake from sleep.

After that I have modified the LowPower example to read a shtc3 sensor over i2c. I checked with my oscilloscope and the STM32 is not sending anything over the I2C bus. The Serial.print() works only until the first wake from sleep but at least the board wakes up as expected from sleep. When I run the firmware in debug(by pressing the Start debugging button next to the upload button in Arduino IDE) all issues are gone - I2C works and Serial.print() works after the first wake. 

I am not experienced in STM32 but I think the issue is related to timing. I have read many issues by other users also in the MbedOS repo and users also report Serial.print() not working after wake.

I am also not able to flash the sketch from the Arduino IDE on MacOS. I have STM32cubeprog installed and working on my Mac but thats what I get from the Arduino IDE:

```
Sketch uses 110728 bytes (42%) of program storage space. Maximum is 262144 bytes.
Global variables use 8648 bytes (13%) of dynamic memory, leaving 56888 bytes for local variables. Maximum is 65536 bytes.
getopt: illegal option -- o
Terminating...
Failed uploading: uploading error: exit status 1
```

**To Reproduce**
This is the sketch I talk about above:
```
#include <STM32LoRaWAN.h>
#include <STM32LowPower.h>
#include <STM32RTC.h>
#include "Adafruit_SHTC3.h"

Adafruit_SHTC3 shtc3 = Adafruit_SHTC3();
STM32LoRaWAN modem;
/* Get the rtc object */
STM32RTC& rtc = STM32RTC::getInstance();

/* Change this value to set alarm match offset in millisecond */
static const unsigned long TX_INTERVAL = 15000; /* ms */

// Declare it volatile since it's incremented inside an interrupt
volatile int alarmMatch_counter = 0;

/* callback function once the Alarm A matched */
void alarmMatch(void* data) {
  UNUSED(data);
  alarmMatch_counter++;
}

void setup() {

  Serial.begin(115200);
  if (! shtc3.begin()) {
    Serial.println("Couldn't find SHTC3");
    while (1) delay(1);
  }
  Serial.println("Found SHTC3 sensor");

  modem.begin(EU868);
  rtc.begin(true);

  bool connected = modem.joinOTAA(/* AppEui */ "xxx", /* AppKey */ "xxx", /* DevEui */ "xxx");

  if (connected) {

  } else {

    while (true)
      ; /* infinite loop */
  }

  // Configure low power
  LowPower.begin();
  LowPower.enableWakeupFrom(&rtc, alarmMatch);
}

void send_packet() {
  Serial.println("Sending packet...");
  // Prepare payload buffer for binary data
  uint8_t payload[5];  // Adjust size as per your data size requirements
  // Read temperature and humidity
  sensors_event_t humidity, temp;
  shtc3.getEvent(&humidity, &temp);
  // Encode temperature and humidity into binary format
  // Assuming temperature and humidity are float values
  uint16_t tempInt = (uint16_t)(temp.temperature * 10);           // Convert to 1 decimal place
  uint16_t humInt = (uint16_t)(humidity.relative_humidity * 10);  // Convert to 1 decimal place

  payload[0] = (tempInt >> 8) & 0xFF;  // High byte of temperature
  payload[1] = tempInt & 0xFF;         // Low byte of temperature
  payload[2] = (humInt >> 8) & 0xFF;   // High byte of humidity
  payload[3] = humInt & 0xFF;          // Low byte of humidity

  // Set port and begin packet
  modem.setPort(10);
  modem.beginPacket();
  modem.write(payload, sizeof(payload));

  // Check if packet was sent successfully
  if (modem.endPacket() == sizeof(payload)) {
    // Packet sent successfully

  } else {
    // Error sending packet
  }
  if (modem.available()) {
    while (modem.available()) {
      uint8_t b = modem.read();
      // Process response if needed
    }
  }
  Serial.println("Packet sent.");
  Serial.flush();
}

void loop() {

  send_packet();

  LowPower.deepSleep(TX_INTERVAL);

  
}

```

Steps to reproduce the behavior:

For unmodified LowPowerBasic example from LoRaWAN lib:
1. Upload LowPowerBasic example to Wii-E5
2. Board sleeps only around 600ms
3. Serial.print() stops working after first wake

For my sketch:
1. Upload the sketch
2. Board sleeps as expected after adding rtc.begin(true) after modem.begin()
3. I2C and Serial.print() not working

**Expected behavior**
I Expect the code to work fine not only in debug mode.

**Desktop**
 - OS: Mac OS Sonoma 14.5
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.8.0
 - STM32 Low Power: 1.2.5
 - STM32 LoRaWAN: 0.2.0
 - Tools menu settings if not the default: Core logs enabled, Optimize: smallest, Runtime Newlib nano, UART: Enabled(generic serial)
 - Upload method: Drag and drop to STLink V3 Mini's mass storage device

**Board (please complete the following information):**
 - Name: Seeed Wio-E5 Mini based on STM32wle5jc
 - Hardware Revision: v1.0

fpistm, 2024-07-01:
Hi @miroslavpetrov 
Thanks for the detailed description but it seems you talk about several one.
One about an issue already opened (using the workaround begin(true)).
One other about I2C not working after a wake up (and serial)?
And one about a download issue.

Moreover it is not clear which board you select in the menu, is it` LoRa-E5 mini`
About upload method, the drag and drop is not supported by default so which upload method failed?

Here it is for bug related to the core. Please split your issue in the correct way:
Comment about the rtc.begin(true) in the correct issue (https://github.com/stm32duino/STM32LoRaWAN/issues/36)
About low power (i2c/Serial) create a new issue in the LoRaWan library.
About upload method, clean up this issue and detail exactly your setup (selected board in the menu and the upload method selected))

miroslavpetrov, 2024-07-01:
@fpistm Thanks for the quick response. I have mentioned I2C,Serial and Sleep issues at once because I think that they are linked together to the improper set of the clock source for the RTC.I will clean the issue and create new ones as you told me.

fpistm, 2024-07-01:
> improper set of the clock source for the RTC.

As stated in other PR, there is no improper clock source for the RTC, As even the modem.begin() call the reset, we are not able to reproduce on our side this seems happen only with third party board and so hard to explain.