stm32duino/Arduino_Core_STM32 #2415: Failed programming: recipe not found 'program.pattern' in v.2.8.0

brightproject, 2024-07-01:
I'm not entirely sure that this is a bug, but without changing it, I conducted a series of observations and experiments.
Also not entirely sure what is causing the crash, but initially decided to write to this community.
I have been using the `arduino-cli` and [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32) combination for a long time.
Some time ago the `arduino-cli` was updated, but there were no problems.
I usually use Arduino IDE version `1.8.19` and update the `stm32` and `esp32` cores through the board manager.
I could not install the `stm32` core using Arduino IDE version `1.8.19`
Updated the kernel via `arduino-cli` by entering the command

> arduino-cli core install STMicroelectronics:stm32

The `stm32` core has been updated to version `2.8.0` and when loading the firmware onto the board

> arduino-cli upload -b STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F411CE,upload_method=swdMethod -i C:\Users\name\AppData\Local\Temp\arduino\sketches\28137305E41AD2AD2370E09B77B89274\test.ino.bin

got this error
`Error: No ports selected. Please select a port.`
I rolled back to version `2.7.0` - everything works fine on it.
I'm not sure if I can repeat the error on other PCs, but version `2.8.0` doesn't work for me.

fpistm, 2024-07-01:
Could you be more precise on your setup?
Arduino ide version? arduino-cli version? Host OS? ....

brightproject, 2024-07-01:
> Could you be more precise on your setup?
> Arduino ide version? arduino-cli version? Host OS? ....

Yes, sure!
Windows 10 pro, 22H2
Arduino-cli 1.0.1
Arduino IDE 1.8.19

BennehBoy, 2024-07-02:
Same issue.
Arch Linux (currently patched)
Arduino-cli  0.35.3  (Arch package not yet updated to 1.0.1)
Arduino IDE 2.3.2

fpistm, 2024-07-02:
@BennehBoy seems strange with arduino ide2.  It works fine on Linux. Which target you used?

SuppaTenko, 2024-07-02:
I second this issue. Saw recipe not found 'program.pattern' error in logs today. Didn't bother about it for some reason.
Now when i try to flash my MCU via STLink first i get read ECONNRESET in arduino logs.
And when i retry - i get 
No connection established
Compilation error: No connection established

And arduino-cli.exe process is dead.
When i only compile my code without flashing or when i export binaries - it works just fine.
My OS is Win10x64 build 17763
Arduino IDE is 2.3.2
Rolled back stm32 core to 2.7.1, and it works good, so it's something with this update...

fpistm, 2024-07-02:
@BennehBoy , @SuppaTenko 
So you get the issue only to flash the board? 
Can you share your command line please?

fpistm, 2024-07-02:
@BennehBoy  Could you test with arduino-cli 1.0.1. There is a lot a breaking change and probably linked...

BennehBoy, 2024-07-02:
I've manually built arduino-cli 1.0.1 and re-tested:

```
arduino-cli compile
Sketch uses 48532 bytes (37%) of program storage space. Maximum is 131072 bytes.
Global variables use 6108 bytes (29%) of dynamic memory, leaving 14372 bytes for local variables. Maximum is 20480 bytes.

Used library                                     Version Path
SPI                                              1.1.0   /home/bennehboy/.arduino15/packages/STMicroelectronics/hardware/stm32/2.8.0/libraries/SPI
Adafruit GFX Library                             1.11.9  /home/bennehboy/Arduino/libraries/Adafruit_GFX_Library
Adafruit BusIO                                   1.16.1  /home/bennehboy/Arduino/libraries/Adafruit_BusIO
Wire                                             1.0.0   /home/bennehboy/.arduino15/packages/STMicroelectronics/hardware/stm32/2.8.0/libraries/Wire
Adafruit SSD1331 OLED Driver Library for Arduino 1.2.2   /home/bennehboy/Arduino/libraries/Adafruit_SSD1331_OLED_Driver_Library_for_Arduino
SrcWrapper                                       1.0.1   /home/bennehboy/.arduino15/packages/STMicroelectronics/hardware/stm32/2.8.0/libraries/SrcWrapper
USBDevice                                        1.0.0   /home/bennehboy/.arduino15/packages/STMicroelectronics/hardware/stm32/2.8.0/libraries/USBDevice

Used platform            Version Path
STMicroelectronics:stm32 2.8.0   /home/bennehboy/.arduino15/packages/STMicroelectronics/hardware/stm32/2.8.0

arduino-cli upload
Failed programming: recipe not found 'program.pattern'

arduino-cli version
arduino-cli  Version: 1.0.1-arch Commit: 590e73bf0412a30780298415180ee085e8d27a07 Date:
```

fpistm, 2024-07-02:
@brightproject 
Install under Arduino 1.8.19 is now ok. Legacy Arduino IDE doesn't support the host value "all". I've updated the json to provide all supported host value: https://github.com/stm32duino/BoardManagerFiles/commit/0e3e525120767e5214201172dae50c2655eb2f1c

Anyway, other issues appears, some variables values defined in the platform.txt are not expanded. I guess, I will not support the legacy Arduino IDE 1.8.x. Supporting new features and specifications from Arduino is not compatible with it.

About the upload issue, I found the root cause, since `programmer.default` is defined the cli try to use it and does not fallback to upload feature like it is done by the Arudino IDE 2.x.
Will try to find a fix, for testing simply remove the line in the menu of your board in boards.txt which set the `programmer.default `

BennehBoy, 2024-07-02:
> About the upload issue, I found the root cause, since `programmer.default` is defined the cli try to use it and does not fallback to upload feature like it is done by the Arudino IDE 2.x. Will try [...]

Confirmed working, thanks.

SuppaTenko, 2024-07-03:
Oh! Caught it!
[image]

fpistm, 2024-07-03:
@SuppaTenko issue is identified and a workaround exist see my comment: https://github.com/stm32duino/Arduino_Core_STM32/issues/2415#issuecomment-2203131381

Fix will come soon.

commit f9a35fd7 (2024-07-04): fix(programmer): remove stlink programmer

as it is not relevant and avoid upload using arduino-cli.

Fixes #2415
closed 2024-07-04 by fpistm