stm32duino/Arduino_Core_STM32 #2416: Error: Cannot run program "{busybox}"

HugoP, 2024-07-03:
I upgraded from Arduino Core STM32 2.7.1 to 2.8.0 and now getting this error when uploading a sketch.

```bash
{busybox} sh C:\Users\User\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.2.2/stm32CubeProg.sh -i swd -f c:\Users\User\Documents\Arduino\generated_examples\.build/Blink.ino.bin -o 0x0 
Uploading...
java.io.IOException: Cannot run program "{busybox}": CreateProcess error=2, The system cannot find the file specified
```

Sketch

```arduino
#include <Arduino.h>

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
    digitalWrite(LED_BUILTIN, HIGH);
    delay(1000);
    digitalWrite(LED_BUILTIN, LOW);
    delay(1000);
}
```

**Desktop**
- OS: Windows 10
- Arduino IDE version: 1.8.19
- STM32 core version: 2.8.0
- Upload method: ST-Link SWD

**Board**
- Custom based on Blue pill with STM32F103CBT

**Additional context**
 - Looks to me like the core thinks it is on Linux based on trying to execute "stm32CubeProg.sh"

When I switch back to v2.7.1 the error goes away.

fpistm, 2024-07-03:
Hi @HugoP 
Unfortunately, the stm32 core is no more compatible with Legacy Arduino IDE 1.8.x only with the Arduino IDE 2.x.

brickZA, 2024-07-14:
I noted a similar issue (`Cannot run program "{busybox}"`) when using STM32CubeProgrammer for uploading with arduino 1.8.13 when upgrading to v2.8.0: https://github.com/stm32duino/Arduino_Core_STM32/issues/2438 

It was pretty easy to fix and my moderately complex sketch kept working, see  https://github.com/stm32duino/Arduino_Core_STM32/pull/2439. However, I don't see an obvious config change related to ST-LINK in `platform.txt` so I'm not sure how to fix this error :)

No harm in trying my fix locally. I found where my STM32 core was installed, edited the `platform.txt` there, and after restarting arduino was off to the races.

diyelectromusic, 2024-07-19:
> Hi @HugoP Unfortunately, the stm32 core is no more compatible with Legacy Arduino IDE 1.8.x only with the Arduino IDE 2.x.

Hi there.

Is this a deliberate roadmap choice to drop support for IDE v1.8.x?  If so then the Getting Started pages appear to differ (I've spent most of the afternoon trying to work out why your instructions weren't working - all the screenshots show v1 series dialogs and there is no mention of requiring v2).

Or is this a bug that is likely to get fixed?

Many thanks,
Kevin

fpistm, 2024-07-19:
I will update the wiki as well. The Readme and release note mention it.
Arduino specifications changed and maintain/test both would be hard so support was drop.

closed 2024-07-20 by fpistm