stm32duino/Arduino_Core_STM32 #2434: Generic STM32F407ZGTx and Serial4 issue

luca-stm32, 2024-07-12:
**Describe the bug**
With STM32F407ZGTx variant, when I declare HardwareSerial Serial4(PC11, PC10), I have a compilation error:

/tmp/arduino/cores/f30dbcaeee57717ef9aa6164ec08b0b8/core.a(HardwareSerial.cpp.o):/home/laboratorio/.arduino15/packages/STMicroelectronics/hardware/stm32/2.8.0/cores/arduino/HardwareSerial.cpp:55: multiple definition of `Serial4'; /tmp/arduino/sketches/AD29A1BE2C5F36AFCFC098E027D51DA2/sketch/Test.ino.cpp.o:/home/laboratorio/Arduino/Test/Test.ino:11: first defined here
collect2: error: ld returned 1 exit status

**To Reproduce**
Use this code:

//                      RX    TX
HardwareSerial Serial1(PA10, PA9);

//                      RX    TX
HardwareSerial Serial2(PA3, PA2);

//                      RX    TX
HardwareSerial Serial4(PC11, PC10);

//                      RX    TX
HardwareSerial Serial5(PD2, PC12);

//                      RX    TX
HardwareSerial Serial6(PC7, PC6);

void setup() {

	/* Serial Configuration */
	Serial1.begin(38400);
	Serial2.begin(115200);
	Serial4.begin(38400);
	Serial5.begin(115200);
	Serial6.begin(115200);

  while(1) {
    delay(250);
    Serial4.print("Test");
  }

}

void loop() {

}

Steps to reproduce the behavior:
Board part number: Generic F407ZGTx
USB Support: NONE
USART Support: Enabled (generic Serial)

**Expected behavior**
Compilation without errors.

**Desktop (please complete the following information):**
 - OS: Linux
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.8

**Additional context**
If I comment HardwareSerial Serial4(PC11, PC10); line, compilation is ok.

I used this code with 2.3.0 and it worked (with Sloeber, but I think that IDE is not so important for this issue).

fpistm, 2024-07-12:
Serial4 is the default Serial instance. So this is normalas you declare it.

closed 2024-07-12 by fpistm

fpistm, 2024-07-15:
To go further, default configuration is defined here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/b24801b4b473649fb6d5bc51c22a69a64d45b732/variants/STM32F4xx/F407Z(E-G)T_F417Z(E-G)T/variant_generic.h#L232-L245

So simply remove the `Serial4 `declaration from your code:
`// HardwareSerial Serial4(PC11, PC10);`
Then if you want to change the default pins used see https://github.com/stm32duino/Arduino_Core_STM32/wiki/API#change-default-serial-instance-pins

```c++
  Serial4.setRx(PC11);
  Serial4.setTx(PC10);
  Serial4.begin(38400);
```