stm32duino/Arduino_Core_STM32 #2436: Maple upload broken

maxgerhardt, 2024-07-13:
**Describe the bug**
When selecting a Bluepill STM32F103C8 and the Mapple Bootloader 2.0 [from here](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/binaries/generic_boot20_pc13.bin), the STM32Duino core fails to use the correct USB PID/VID of 0x1EAF:0x003 and instead uses 0x0483:0xdf11. Upload subsequently fails to find the correct device.

**To Reproduce**
No source code relevant. Blinky sketch for PC13.

Steps to reproduce the behavior:
1. Use ST-Link or USB-to-UART adapter to first mass erase the flash
2. Burn in generic_boot20_pc13.bin
3. Reboot the Bluepill, "Maple DFU" device shows up (asssumes [driver is installed](https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/drivers)
4. Select Bluepill C8 and Maple Bootloader 2.0 in the Tool settings
5. Hit upload
6. See an invocation of 
```
"C:\Users\Max\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.2.2/win/busybox.exe" sh "C:\Users\Max\AppData\Local\Arduino15\packages\STMicroelectronics\tools\STM32Tools\2.2.2/maple_upload.sh" COM1 2 0x0483 0xdf11 "C:\Users\Max\AppData\Local\Temp\arduino\sketches\68F814A2F7381590CAB9CFF90ED55329/sketch_jun16a.ino.bin"
```

Get hit with failure

```
No DFU capable USB device found
```
(no wonder if you select teh wrong VID/PID)

**Expected behavior**
Uploading with the Maple bootloader **actually** uses the right PID/VID

**Screenshots**
None

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.8.0
 - Tools menu settings if not the default: per
[image]
 - Upload method: Maple Bootloader 2.0

**Board (please complete the following information):**
 - Name: Bluepill STM32F103C8
 - Hardware Revision: -/-
 - Extra hardware used if any: None

**Additional context**
Add any other context about the problem here.

maxgerhardt, 2024-07-13:
And wtf is this doing

https://github.com/stm32duino/Arduino_Core_STM32/blob/b24801b4b473649fb6d5bc51c22a69a64d45b732/boards.txt#L2526-L2528

It spells `0xLEAF`  but `L` for sure is not a hexadecimal number, it should be `1EAF`

maxgerhardt, 2024-07-13:
Oh lookie lookie there, when I change

https://github.com/stm32duino/Arduino_Core_STM32/blob/b24801b4b473649fb6d5bc51c22a69a64d45b732/platform.txt#L113-L114

to be 
```
upload.vid=0x1eaf
upload.pid=0x0003
```
I do get a successfull upload

[image]

maxgerhardt, 2024-07-13:
So obviously, when Tools -> Upload Method --> Maple DFU Bootloader 2.0 is selected, it should use 1EAF:0003 as the VID:PID for DFU-Util. This is probably even irregardless of the selected chip (doesn't matter whether "Generic STM32F103C8" or "Bluepill C8", since the Maple DFU bootloader will **always** show up as that VID:PID. I'm not sure what this core here is doing using that other VID:PID, because that is the generic "STM Device in DFU Mode" VID/PID, **not** the one for Maple.

[image]

I'm a bit baffled that the most popular Bluepill board with the most popular USB bootloader can't be used in this core out of the box-

fpistm, 2024-07-15:
Hi @maxgerhardt 
Thanks for reporting this

> And wtf is this doing

A typo mistake, sorry for this. I've made a lot of changes to feat new Arduino specification. I've tested the Blue Pill but when rebased and squashed several branches I've made a mistake. I'm only human so please be comprehensive....

> I'm a bit baffled that the most popular Bluepill board with the most popular USB bootloader can't be used in this core out of the box-

Same here, this kind of comment is really not constructive and only your opinion. It is a community project and all contributions are welcome (PR, review, discussions,...). Doing my best (alone) ...

Fixed by #2440.

PR #2440 (2024-07-15): fix(upload_vid): wrong definitions
Fixes #2436
closed 2024-07-15 by fpistm