stm32duino/Arduino_Core_STM32 #2438: Error uploading with stm32CubeProgrammer (DFU) since v2.8.0

brickZA, 2024-07-14:
**Describe the bug**
Get the following error uploading to a F401CCx blackpill board when upgrading to v2.8.0:

```
Sketch uses 39208 bytes (14%) of program storage space. Maximum is 262144 bytes.
Global variables use 5416 bytes (8%) of dynamic memory, leaving 60120 bytes for local variables. Maximum is 65536 bytes.
java.io.IOException: Cannot run program "{busybox}": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2055)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch
```

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
1. Upgrade core to v2.8.0
2. Upload as per usual
3. See error above

**Expected behavior**
It uploads succesfully

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 x86-64
 -  - Arduino IDE version: 1.8.13
 - STM32 core version: 2.8.0
 - Tools menu settings if not the default:
   - Board: Generic STM32F4 series
   - Board part number: Generic F401CCUx
   - U(S)ART support: Enabled (generic 'Serial')
   - USB support: CDC (generic 'Serial' supersede U(S)ART
   - Upload method: STM32CubeProgrammer (DFU)  
 -  [e.g. Newlib Standard, No Serial]
 - Upload method: [e.g. SWD]

**Board (please complete the following information):**
 - Name: STM32F401 Black Pill Board with STM32F401CCU6
 
**Additional context**

I managed to fix it by noting some changes in the `platform.txt` between v2.7.1 (works) and v2.8.0 (broken) relating to the `busybox` variable and the `stm32CubeProgrammer` section. If I apply the following diff the upload works without any issues:

```
diff --git a/platform.txt b/platform.txt
index 3b60a337c..9e6cc7171 100644
--- a/platform.txt
+++ b/platform.txt
@@ -198,6 +198,8 @@ tools.massStorageCopy.upload.params.quiet=
 tools.massStorageCopy.upload.pattern="{tools_bin_path}/{cmd}" {upload.verbose} -I "{build.path}/{build.project_name}.bin" -O "{node}"
 
 # STM32CubeProgrammer upload
+tools.stm32CubeProg.busybox=
+tools.stm32CubeProg.busybox.windows={path}/win/busybox.exe
 tools.stm32CubeProg.path={runtime.tools.STM32Tools.path}
 tools.stm32CubeProg.cmd=stm32CubeProg.sh
 tools.stm32CubeProg.upload.params.verbose=
```

If I can figure how I'll make a PR to fix :)

brickZA, 2024-07-14:
Created a PR with above diff: https://github.com/stm32duino/Arduino_Core_STM32/pull/2439

fpistm, 2024-07-15:
Hi @brickZA 
Core is no more compatible with Arduino IDE 1.8.x. Arduino specifications got a lot of changed and new ones so we do not support anymore the 1.8.x ide.

closed 2024-07-15 by fpistm