stm32duino/Arduino_Core_STM32 #2441: Wire memory leak on multiple begin() call

benmalenfant, 2024-07-16:
**Describe the bug**
When  calling begin after the recieve and transmit buffer have been allocated, the begin function orphans the allocation by resetting the pointer to nullptr, on the next call of allocateTxBuffer(), we get a new memmory allocation while the previous one has never been freed, this in turns causes heap segmentation and will lead to lack of memmory in the long run.

**To Reproduce**
Using this simplified code, we can see that the Wire recieve buffer is reallocated every requestFrom call
```
#include <Arduino.h>
#include <Wire.h>

void setup() {
}

void loop() {
  Wire.begin();
  Wire.requestFrom(1,1);
}
```

Steps to reproduce the behavior:
1. Run demo code
2. Place breakpoint in allocateRxBuffer right after the realloc line
3. Notice the tmp pointer increase location every call
4. Wait long enough and you'll run out of memmory

**Expected behavior**
The wire library should not cause memory leak, even if calling begin() multiple time is bad practice.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Board (please complete the following information):**
 - Name: BlackPill STM32F411CEU6
 - Hardware Revision: V3.1

 Applicable to any stm32

**Additional context**
Will open a pull request soon with a proposed fix

fpistm, 2024-07-16:
Hi. In fact your code is wrong. You have to call `Wire.end()`;
This is like you call malloc each loop iteration without calling free.

benmalenfant, 2024-07-16:
Yeah I agree, this is dummy code, but my philosophy is that Arduino should be dummy proof. This became an issue for me because I had locally scoped a sensor object(I know this was stupid...), destroying and creating it every time caused it to call begin and in turn cause the memory leak.

PR #2442 (2024-07-16): Fix Wire memory leak
**Summary**

This PR fixes the memory leak bug caused by multiple Wire.begin() calls, mentioned in #2441

The strategy used here is to initialise the buffer variables on the Wire object initialisation instead of the begin.

**Validation**

* Ensure CI build is passed.
* Ran the example code in #2441, now allocateRxBuffer() is only called once at the first call of recieveFrom() and the memory leak has been fixed

**Code formatting**

* Ensure AStyle check is passed thanks CI

**Closing issues**

Fixes #2441
closed 2024-07-16 by fpistm