stm32duino/Arduino_Core_STM32 #2443: SPI support to other libraries using  TRANSMIT ONLY or RECEIVE ONLY

AndreoBotelho, 2024-07-16:
**Describe the bug**
stm hang when trying to send data to SPI bus

**To Reproduce**
`
#include <SPI.h>

#define TFT_MOSI PE14
#define TFT_SCLK PE12
#define TFT_CS   PE4  // Chip select control pin
#define TFT_DC   PE13  // Data Command control pin
#define TFT_RST  PE11  // Reset pin (could connect to RST pin)
#define TFT_BL   PC7  // bl pin (could connect to bl pin)

void setup() {
  Serial.begin(115200);
  pinMode(TFT_CS, OUTPUT);
  pinMode(TFT_DC, OUTPUT);
  pinMode(TFT_BL, OUTPUT);
  digitalWrite(TFT_BL, LOW);  
  digitalWrite(TFT_CS, LOW);  
  digitalWrite(TFT_DC, LOW);  
  SPI.setMOSI(TFT_MOSI);
  //SPI.setMISO(PE5);
  SPI.setSCLK(TFT_SCLK);
  SPI.setClockDivider(2);
  SPI.begin();
  Serial.println("Inited");
}

// the loop function runs over and over again forever
void loop() {

  Serial.println("Loop");
  digitalWrite(TFT_BL, HIGH);  // turn the LED on (HIGH is the voltage level)
  digitalWrite(TFT_CS, LOW);  
  SPI.transfer(0x21, true);
  digitalWrite(TFT_CS, HIGH);  
  delay(1000);                      // wait for a second
  digitalWrite(TFT_CS, LOW);  
  digitalWrite(TFT_BL, LOW);   // turn the LED off by making the voltage LOW
  SPI.transfer(0x28, true);
  digitalWrite(TFT_CS, HIGH);  
  delay(1000);  // wait for a second
}`

Steps to reproduce the behavior:
1. power on

**Expected behavior**
send bytes to display

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS:  Windows
 - Arduino IDE version: 2.3.2
 - STM32 core version: latest
 - Tools menu settings if not the default: default
 - Upload method: DFU

**Board (please complete the following information):**
 - Name: STM32H743 custom board
 - Hardware Revision: none
 - Extra hardware used if any: SPI ST7789 display

**Additional context**
 SPI.setClockDivider(int); is not working, I can get only 256 divider, manually coding 2 divider works fine
uncommenting //SPI.setMISO(PE5); make it works but I don't have this MISO pin connected and loses a GPIO on PE5

are there any plans to support DMA transfer? the current code is slow and eats a lot of cpu power.

fpistm, 2024-07-16:
Hi @AndreoBotelho 
about setClockDivider it should not be used, you have to use SPISettings:

fpistm, 2024-07-16:
About your issue with clock divider did you check the clock config? 
About the need of miso, yes it is required to properly init the SPI:
https://github.com/stm32duino/Arduino_Core_STM32/blob/3a3ff8428655c0fb314a48b06461a6b78a2833f7/libraries/SPI/src/utility/spi_com.c#L222-L226

About DMA, it is not planned. Anyway, feel free to contribute 😉

I see no specific issue (except maybe the clock divider) so what did you expect?