stm32duino/Arduino_Core_STM32 #2445: 2.8.0 breaks custom variants

mtboswell-aeroseal, 2024-07-16:
**Describe the bug**
When compiling an Arduino sketch for a custom variant defined in the sketchbook's /hardware/ folder, the SrcWrapper library is not found because the path to the SrcWrapper library is specified with the {runtime.platform.path} variable which points to the hardware folder in the sketchbook instead of the hardware folder in the ST package.

**To Reproduce**

Steps to reproduce the behavior:

1. Create a variant based on the STM32 Arduino platform and place it in the <sketchbook>/hardware/<name>/stm32/ folder. 
2. Attempt to compile an empty sketch.

**Expected behavior**
Libraries are found and sketch compiles.

**Screenshots**
None

**Desktop (please complete the following information):**
 - OS: Windows
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.8.0
 - Tools menu settings if not the default: N/A
 - Upload method: N/A

**Board (please complete the following information):**
 - Name: Custom
 - Hardware Revision: Custom
 - Extra hardware used if any: N/A

**Additional context**
Line 32 of platform.txt changed the path of several includes to point to a path including the {runtime.platform.path} variable. This variable does not always point to the ST board package and thus breaks the include path for custom variants specified in the sketchbook folder.

fpistm, 2024-07-16:
Have you an zip to reproduce this? I don't understand what you called a custom variant? Or any Arduino documentation ?

mtboswell-aeroseal, 2024-07-16:
Unfortunately, we do not have a zip that we can share publicly. 

Here is the part of the platform specification that describes referencing another platform: https://arduino.github.io/arduino-cli/1.0/platform-specification/#referencing-another-core-variant-or-tool .

In our %SKETCHBOOK%\hardware\MyOrg\stm32\boards.txt file we have a line similar to:

myboard.build.core=STMicroelectronics:arduino

Which allows us to use the ST core even though we have our own separate boards.txt.

This works fine on 2.7.1, but on 2.8.0 the compile fails because the ST include path now has the {runtime.platform.path} variable in it, and the {runtime.platform.path} variable changes from "%LOCALAPPDATA%\Arduino15\packages\STMicroelectronics\hardware\stm32\2.8.0" to "%SKETCHBOOK%\hardware\MyOrg\stm32" when a board from the local platform is selected.

mtboswell-aeroseal, 2024-07-16:
Allow me to make this easier. We were able to solve the problem by replacing lines 31-33 of your platform.txt with:
```
SrcWrapper_include_dir={build.core.path}/../../libraries/SrcWrapper/inc
VirtIO_include_dir={build.core.path}/../../libraries/VirtIO/inc
USBDevice_include_dir={build.core.path}/../../libraries/USBDevice/inc
```

mtboswell-aeroseal, 2024-07-16:
Also, if a board listing does not specify build.vid and build.pid it will also not compile because there are no valid default values for those parameters.

fpistm, 2024-07-17:
> Also, if a board listing does not specify build.vid and build.pid it will also not compile because there are no valid default values for those parameters. The comment on line 110 of platform.txt says that we should be able to omit them, but this is not the case.

About default vid/pid, it is mentionned in this commit message:
https://github.com/stm32duino/Arduino_Core_STM32/pull/2323/commits/8d9c950748ae28610b60f171913d76649a34dff6

> Note that vid.0/pid.0 have to be defined for each menu even if no USB
> support to have the build.vid/pid properly defined in the platform.txt.
> At least a FTDI is defined (ex: STLink VCP)
> 

I agree the message was not updated in platform.txt. I will update it or try to define a default ones. 
So simply defined them with the STLink vid/pid like this if you used one then COM port will be recognized.

closed 2024-07-18 by fpistm

fpistm, 2024-07-18:
Hi  @mtboswell-aeroseal 

About:
> Also, if a board listing does not specify build.vid and build.pid it will also not compile because there are no valid default values for those parameters. The comment on line 110 of platform.txt says that we should be able to omit them, but this is not the case.

I've made a PR to fix it: #2454

PR #2449 (2024-07-18): fix(platform): core reference broken
due to {runtime.platform.path} usage when other platform reference the core.

- {runtime.platform.path}:
  The absolute path of the board platform folder (i.e. the folder containing boards.txt)
- {build.core.path}:
  The path to the selected board's core folder (inside the  core platform, for example hardware/STMicroelectronics/stm32/core/arduino)
    
Ref:
  https://arduino.github.io/arduino-cli/1.0/platform-specification/#referencing-another-core-variant-or-tool
    
Fixes #2445
commit f4a34098 (2024-07-18): fix(platform): core reference broken

due to {runtime.platform.path} usage when other platform
reference the core.

- {runtime.platform.path}:
  The absolute path of the board platform folder
  (i.e. the folder containing boards.txt)
- {build.core.path}:
  The path to the selected board's core folder (inside the
  core platform, for example hardware/STMicroelectronics/stm32/core/arduino)

Ref:
https://arduino.github.io/arduino-cli/1.0/platform-specification/#referencing-another-core-variant-or-tool

Fixes #2445