stm32duino/Arduino_Core_STM32 #2452: J-Link support when using STM32CubeProgrammer

jamesy0ung, 2024-07-18:
STM32CubeProgrammer has support for J-Link probes, however the Arduino core only allows it to use SWD with ST-Link, DFU or UART

I'd like to be able to upload using a J-Link probe from the Arduino IDE. Thank you!

[image]

fpistm, 2024-07-18:
Hi @jamesy0ung 
Feel free to contribute and provide a PR.

jamesy0ung, 2024-07-18:
I don't know much about the internals of Arduino Cores, though I'd be happy to if I can get some pointers as to what needs to be modified

I did a search for STM32CubeProgrammer and saw some references to it in platform.txt, however I'm not sure what exactly creates the option in the Arduino IDE and where/how I can extend it.

I also searched up the docs for the STM32CubeProgrammer CLI and it's just a flag passed to it.

jamesy0ung, 2024-07-18:
I haven't done any work on Arduino cores before, but I got the CLI working by using the following flag. `-c port=JLINK ap=0`

I tried to modify boards.txt to add support but I can't see it in the Arduino IDE as a programmer
```

GenF1.menu.upload_method.jlinkMethod=STM32CubeProgrammer (J-Link)
GenF1.menu.upload_method.jlinkMethod.upload.protocol=swd
GenF1.menu.upload_method.jlinkMethod.upload.options=-c port=JLINK ap=0
GenF1.menu.upload_method.jlinkMethod.upload.tool=stm32CubeProg
```

Is there anything I'm missing?

fpistm, 2024-07-18:
I would recommend to check the script:
https://github.com/stm32duino/Arduino_Tools/blob/main/stm32CubeProg.sh

And simply add a new interface (protocol): jlink then in the script simply call it with correct option.
About the fact it is not in the menu, this is an Arduino IDE issue. You have to delete the cache.
#### Path:
* Linux: `$HOME/.config/arduino-ide/`
* Windows: `C:\Users\<login>\AppData\Roaming\arduino-ide`

jamesy0ung, 2024-07-18:
I've opened 2 PR's

One adds a case to the flashing script, and the other adds support under the STM32F1, I'll look into adding J-Link programmer support for other boards, but I don't have any other hardware to test with

Amir1387aht2, 2024-09-29:
Guys Any New Info? I Wanna Use Jlink Flasher And There's No Straight Option

PR #2455 (2024-10-02): J-Link Support for STM32F1
**Summary**

This PR implements the support for using a STM32CubeProgrammer with a J-Link from the Arduino IDE

Explain the **motivation** for making this change. What existing problem does the pull request solve?

I have a J-Link probe and can use it directly in STM32CubeProgrammer, but not in the Arduino IDE

**Validation**

I tested it and it works, doesn't seem to break anything else.

close #2452
closed 2024-10-02 by fpistm