stm32duino/Arduino_Core_STM32 #2459: Specify `SERIAL_RX_BUFFER_SIZE` or similar define

zfields, 2024-07-21:
`SERIAL_RX_BUFFER_SIZE` is a define from the legacy Arduino Cores.

- https://github.com/arduino/ArduinoCore-avr/blob/321fca0bac806bdd36af8afbc13587f4b67eb5f1/cores/arduino/HardwareSerial.h#L49
- https://github.com/arduino/ArduinoCore-megaavr/blob/5e639ee40afa693354d3d056ba7fb795a8948c11/cores/arduino/UART.h#L50

We use it to optimize large serial transfers, and it would be nice to see it specified in the `Arduino_Core_STM32`.

If you have a different / similar approach, that would also be very helpful.

fpistm, 2024-07-22:
Hi @zfields 

it is documented in the wiki:
https://github.com/stm32duino/Arduino_Core_STM32/wiki/Custom-definitions#serial-rxtx-buffer-size

A simple search point to it.

closed 2024-07-22 by fpistm

zfields, 2024-07-22:
I searched for `SERIAL_RX_BUFFER_SIZE` against the repo, and it didn't return as a result, which is why I raised the issue.

In the wiki it says,

```
By default, Serial Rx/Tx buffer size are defined like this:

#if !defined(SERIAL_TX_BUFFER_SIZE)
#define SERIAL_TX_BUFFER_SIZE 64
#endif
#if !defined(SERIAL_RX_BUFFER_SIZE)
#define SERIAL_RX_BUFFER_SIZE 64
#endif
```

In which file are these defined (by default)?