stm32duino/Arduino_Core_STM32 #2460: STM32Duino on F103ZET6 ignores -DHAL_TIM_MODULE_ONLY in build_opt.h

vtpgit, 2024-07-23:
**Describe the bug**

STM32Duino on F103ZET6 ignores -DHAL_TIM_MODULE_ONLY in build_opt.h
The build_opt.h is properly processed for the F103C8ET6 however.

I'm building the basic demo from BlueVGA library.

[image]

[image]

[image]

For a generic F103C8ET6 the build completed:

[image]

**To Reproduce**

Steps to reproduce the behavior:

Open the example simple_demo from the library.

Select F103C8ET6 as a target.

Build. The build completes.

Select F103ZET6 as a target.

Select F103C8ET6 as a target.

Build. The build fails with redeclaration error, pointing to ignoring the directive HAL_TIM_MODULE_ONLY in build_opt.h
as defined in 

https://github.com/stm32duino/Arduino_Core_STM32/wiki/HAL-configuration

[image]

**Expected behavior**
STM32Duino on F103ZET6 must honor -DHAL_TIM_MODULE_ONLY in build_opt.h directive.

Please let us know what could be done meanwhile to work-around. We understand such fixes take time.

**Screenshots**

Attached above.

**Desktop (please complete the following information):**
 - OS:  Windows Server 2019
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.8.1
 - Tools menu settings : screenshot provided
 - Upload method: build fails, doesn't matter

**Board (please complete the following information):**
 - Name: STM32F103ZET6 Generic

Not really relevant but for the record.

[image]

**Additional context**

Assumes BlueVGA is pre-loaded.

Code for sample:

```
#include "bluevga.h"
#include "font.h"

// creates a BlueVGA object using ASCII_FONT as bitmap for all the tiles (ASCII characters)
BlueVGA vga(ASCII_FONT);

void setup() {
  ScreenSetup();    // performs initial screen character drawing
  Animation();      // forever loop...
}

void loop() {
  // no need to code anything here... it's just a function that is called for ever
  // setup() can do it all, with a for_ever_loop such as in Animation()
}

/*
    Performs a simple color animation
*/
void Animation (void) {

  static uint8_t lastLineColor = 1;
  // defines an order and sequence of colors for the animation
  const uint8_t colors[8] = {RGB_RED, RGB_YELLOW, RGB_GREEN, RGB_CYAN, RGB_BLUE, RGB_MAGENTA, RGB_BLACK, RGB_WHITE};

  while (true) {   // forever ... sort of replaces loop()

    // the animation is excecuted 2 times per second...
    vga.waitVSync(30); // blocks the execution until 30 frames are past. At 60 FPS (frames per second) 30/60s = 500 milliseconds

    // gets the current Foreground and Backgorund colors for a specific position in screen
    // in particular, this position is the top left cornner of the ASCII chart at screen
    
    uint8_t x = (VRAM_WIDTH - 16) / 2;
    uint8_t fc = vga.getFGColor(x, 3);
    uint8_t bc = vga.getBGColor(x, 3);

    // rotates foreground and background colors.
    bc += 2; // colors are 4 bits, but only 3 most significative bit are used, thus it is always a pair number
    // Background goes from 0 to 14 and then foreground changes
    if (bc > 14) { // it goes up to 14 (0xE) = white
      bc = 0;
      fc += 2;
    }
    if (fc > 14) fc = 0;

    // changes ASCII chart colors
    vga.setColorRegion(x, 3, x + 16 - 1, 8, vga.getColorCode(fc, bc));

    // prints "last line" at the bottom of the screen and rotates its color as defined in the array colors[]
    lastLineColor = ++lastLineColor & 7; // increments and keeps the range in 0..7 for a single color index in the sequence we defined
    x = (VRAM_WIDTH - 13) / 2;
    vga.setColorRegion(x, 29, x + 13, 29, vga.getColorCode(colors[lastLineColor], 0)); // foreground, background colors

    // prints a number (color code) with leading '0's and 2 digits
    vga.printInt (24, 29, colors[lastLineColor], vga.getColorCode(RGB_BLACK, RGB_WHITE), true, 2);
  }
}

void ScreenSetup (void) {

  vga.clearScreen();
  vga.printStr(0, 20, vga.getColorCode(RGB_YELLOW, RGB_BLACK), (char *)"12345678901234567890123456789012");

  for (uint8_t i = 0; i < VRAM_WIDTH; i++) {  // draws character #127 (checker) in 3 rows in the screen, alternating colors, keeping Red as main color
    // there are 3 ways for drawing a single character in the screen, using setTile(...):
    vga.setTile(i, 22, 127, vga.getColorCode(RGB_RED, RGB_YELLOW));   // using setTile(...) with a color at last paramenter
    vga.setTile(i, 23, 127, RGB_BLACK, RGB_RED);                      // using setTile(...) with arguments separated for foreground and background colors
    vga.setTile(i, 24, 127);                                          // using setTile(...) with no color argument, thus it doesn't change the colors of that x,y position
    vga.setColor(i, 24, vga.getColorCode(RGB_RED, RGB_WHITE));        // then it may be possible to only change the color at x,y not changing the tile at that position
  }

#ifdef ARDUINO_ARCH_STM32F1  // Roger's BluePill Core https://github.com/rogerclarkmelbourne/Arduino_STM32
  vga.printStr((VRAM_WIDTH - 18) / 2, 0, vga.getColorCode(RGB_YELLOW, RGB_BLACK), (char *)"Roger's Core DEMO!");
#endif

#ifdef ARDUINO_ARCH_STM32  // Arduino_Core_STM32 Core https://github.com/stm32duino/Arduino_Core_STM32
  vga.printStr((VRAM_WIDTH - 16) / 2, 0, vga.getColorCode(RGB_YELLOW, RGB_BLACK), (char *)"STM32 Core DEMO!");
#endif

  // prints all possible colors using text in the screen
  uint8_t xCenter = (VRAM_WIDTH - 14) / 2;
  vga.printStr(xCenter, 11, vga.getColorCode(RGB_RED, RGB_BLACK), (char *)    "-----RED------");
  vga.printStr(xCenter, 12, vga.getColorCode(RGB_MAGENTA, RGB_BLACK), (char *)"---MAGENTA----");
  vga.printStr(xCenter, 13, vga.getColorCode(RGB_BLUE, RGB_BLACK), (char *)   "-----BLUE-----");
  vga.printStr(xCenter, 14, vga.getColorCode(RGB_CYAN, RGB_BLACK), (char *)   "-----CYAN-----");
  vga.printStr(xCenter, 15, vga.getColorCode(RGB_GREEN, RGB_BLACK), (char *)  "----GREEN-----");
  vga.printStr(xCenter, 16, vga.getColorCode(RGB_YELLOW, RGB_BLACK), (char *) "----YELLOW----");
  vga.printStr(xCenter, 17, vga.getColorCode(RGB_WHITE, RGB_BLACK), (char *)  "----WHITE-----");
  vga.printStr(xCenter, 18, vga.getColorCode(RGB_BLACK, RGB_WHITE), (char *)  "BLACK REVERSED");
  vga.printStr(xCenter, 29, vga.getColorCode(1, RGB_BLACK), (char *)  "--Last Line--");

  // prints the basic ASCII table in the screen
  for (uint8_t y = 3; y < 9; y++)
    for (uint8_t x = (VRAM_WIDTH - 16) / 2; x < (VRAM_WIDTH - 16) / 2 + 16; x++) {
      vga.setTile(x, y, ' ' - (VRAM_WIDTH - 16) / 2 + x + (y - 3) * 16);
      vga.setColor(x, y, vga.getColorCode(RGB_WHITE, RGB_BLACK));
    }
}

```

fpistm, 2024-07-23:
Hi @vtpgit,
As stated by the BlueVGA readme, it is only compatible with **Generic STM32F103C Series**:
https://github.com/RoCorbera/BlueVGA/blob/master/README.md#how-to-use-it-with-arduino-ide

You try to build for a` Generic STM32F103Z Series`, error is normal as stated by your log. It includes 2 CMSIS devices header while only one should be used. 
This file is linked to the device you build but the library hardly includes it:
https://github.com/RoCorbera/BlueVGA/blob/47d8217cd2357afc12d0afd3a409602ebb63be11/src/bluevgadriver.c#L45

So there is no issue with the core not the library as it clearly specify which target is compatible. If you want to support it then up to you to update the library but it is out of scope here.

closed 2024-07-23 by fpistm

vtpgit, 2024-07-23:
> So there is no issue with the core not the library as it clearly specify which target is compatible. If you want to support it then up to you to update the library but it is out of scope here.

@fpistm 

Hi, thank you for your response, I didn't notice that, and updated the library, it built just fine.

I'll be more careful in the future.

Cheers,
Valentine