stm32duino/Arduino_Core_STM32 #2462: 2.8.1 Update disabled Serial() on Nucleo H503

LightningStalker, 2024-07-23:
**Describe the bug**
No serial debug through STLINK3 onboard

**To Reproduce**
Serial worked fine, update to 2.8.1, upload code and no serial, program running, still persist after downgrading to previous core version

Steps to reproduce the behavior:
1. Update core
2. Upload program
3. bye bye Serial.println

**Expected behavior**
Working as before

**Desktop (please complete the following information):**
 - OS: Linux Mint
 - Arduino IDE version: 2.3.2
 - STM32 core version: 2.8.1
 - Tools menu settings if not the default: default
 - Upload method: SWD

**Board (please complete the following information):**
 - Name: Nucleo H503
 - Extra hardware used if any: no

**Additional context**
LED fades running program

LightningStalker, 2024-07-23:
Probing USART1 and USART3 there is no TX from VCP. Missing VCP.

fpistm, 2024-07-23:
HI @LightningStalker 
> Probing USART1 and USART3 there is no TX from VCP. Missing VCP.

I don't understand this? 
Is the board modified?

closed 2024-07-23 by fpistm

LightningStalker, 2024-07-24:
Very sorry I had accidentally selected Generic variant.