stm32duino/Arduino_Core_STM32 #2468: I2C not working as intended for addresses starting with 0b1111 0xxx

blazej222, 2024-08-01:
**Describe the bug**
When STM32 is I2C master, any `beginTransmission(x)` call for x like `0b1111 0xxx` without `write()` call in between will cause subsequent `endTransmission()` call to return `I2C_NACK`, despite slave device returning ACK.

This is due to the fact that STM32 automatically detects addresses starting with `0b1111 0xxx` as 10 bit addresses, and in that case, does not set `I2C_FLAG_ADDR` flag which is being checked after sending the first byte of address.

https://github.com/stm32duino/Arduino_Core_STM32/blob/5e4f22065273339bbd74c01b8209e9592d2c30bd/system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_i2c.c#L3578-L3605

**To Reproduce**

I2C master code:
```cpp
#include <Arduino.h>
#include <Wire.h>

HardwareSerial Serial1(PA10,PA9);
 
void setup() {
    Serial1.begin(115200);
    Wire.begin();    

}
 
void loop() {
        Wire.beginTransmission(0x79); 
        //Wire.write(1); //uncomment to receive ACK, but receive garbage on the other end
        int result = Wire.endTransmission();   
        Serial1.println(result);
        delay(1000);
 }
```

I2C slave code:
```cpp
#include <Arduino.h>
#include <Wire.h>

HardwareSerial Serial1(PA10,PA9); //remove when testing on Arduino

void handleData(int bytes){
  byte data = Wire.read();
  Serial1.println(data); //change to Serial if testing on Arduino
}

void setup() {
  Serial1.begin(115200);  //change to Serial if testing on Arduino
  Wire.begin(0x79);
  Wire.onReceive(handleData);
  Serial1.println("Starting");  //change to Serial if testing on Arduino
}

void loop() {
}

```

Steps to reproduce the behavior:
1. Connect STM32F103C8T6 (I2C Master) to another STM32F103C8T6/Arduino (I2C slave). (GND,SDA,SCL

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: [e.g. 1.8.8]
 - STM32 core version: [e.g. 1.4.0]
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: [e.g. SWD]

**Board (please complete the following information):**
 - Name: [e.g. Nucleo F429ZI]
 - Hardware Revision: [e.g. Rev B]
 - Extra hardware used if any: [e.g. X-Nucleo IKS01A1]

**Additional context**
Add any other context about the problem here.

fpistm, 2024-08-14:
Hi @blazej222 
For me ther is no error:
https://www.i2c-bus.org/addressing/

The address you used rely on the 10 bits addressing:  `1111 0xxx`

Moreover this is the HAL and hardware which handle that so nothing could be done here.

blazej222, 2024-08-16:
My apologies, I might have worded this issue badly. 
Let's assume we have a slave device on I2C that has a 10bit address of `0x208` (`010 0000 1000`). 
To communicate with the device, the following two bytes must appear on I2C bus (as seen with oscilloscope): `0xF2 0x08` (`1111 0010 0000 1000`).

Normally on arduino, you can communicate with this device simply by doing something like this:
```cpp
Wire.begin(0x79);
Wire.write(0x08);
//insert additional communication here
Wire.endTransmission()
```

However, when using the same code on STM32Duino this communication won't succeed.
First of all, `0xF2` (`0x79 << 1`) byte will appear twice on the bus (because we are configuring HAL's I2C interface in 7bit mode, however the MCU itself recognizes this address as the beginning of 10 bit address).
As 10bit mode uses different flags than 7bit mode(for example, it doesn't set the `I2C_FLAG_ADDR` flag HAL's checking for) it produces glitchy results in the communication.

As a result, there's no proper way to communicate with devices that use 10bit addressing from STM32duino. 

The problem itself isn't present in HAL library - on native STM32 cube all you have to do is configure the interface into `I2C_ADDRESSINGMODE_10BIT` and the communication will be handled properly.

I understand that Arduino library does not explicitly provide 10bit addressing support - however, on native Arduino library it's possible to communicate with 10bit I2C devices by simply splitting the address into two bytes.
In case of STM32duino it's not possible to communicate at all.

Probably a check would have to be implemented in STM32duino that configures HAL into 10bit addressing mode when applicable.

fpistm, 2024-08-19:
Hi @blazej222 
OK thanks for the feedback so your request is to have 10 bits address support.
Currently, the Wire library support only 7 bits address.

PR #2493 (2024-08-22): fix(wire): support only 7 bits addressing mode
Arduino does not support 10 bits addressing mode.
The API is limited to an `uint8_t` address:
https://github.com/arduino/ArduinoCore-API/blob/4a02bfc0a924e1fec34c3bb82ffd5dfba7643a0c/api/HardwareI2C.h#L31
https://github.com/arduino/ArduinoCore-API/blob/4a02bfc0a924e1fec34c3bb82ffd5dfba7643a0c/api/HardwareI2C.h#L36

About the hack to use 10 bits on Arduino:

> Normally on arduino, you can communicate with this device simply by doing something like this:
> 
> ```C++
> Wire.begin(0x79);
> Wire.write(0x08);
> //insert additional communication here
> W [... cut]
commit 1b227dbc (2024-08-22): fix(wire): support only 7 bits addressing mode

Fixes #2468.
closed 2024-08-22 by fpistm