stm32duino/Arduino_Core_STM32 #2475: LOAD segment with RWX permissions during stm32f411ceu6 flashing on arduinoststm32 @ 4.20801.240802 (2.8.1)

brightproject, 2024-08-11:
The error is not critical, because the program code works without any comments, but I don't like the concept itself, when I see warnings during compilation:

> Executing task in folder Example: C:\Users\test\.platformio\penv\Scripts\platformio.exe run 
> 
> Processing genericSTM32F411CE (platform: ststm32; board: blackpill_f411ce; framework: arduino)
> -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/blackpill_f411ce.html
> PLATFORM: ST STM32 (17.4.0) > WeAct Studio BlackPill V2.0 (STM32F411CE)    
> HARDWARE: STM32F411CEU6 100MHz, 128KB RAM, 512KB Flash
> DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
> PACKAGES:
>  - framework-arduinoststm32 @ 4.20801.240802 (2.8.1)
>  - framework-cmsis @ 2.50900.0 (5.9.0)
>  - toolchain-gccarmnoneeabi @ 1.120301.0 (12.3.1)
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf

> Compiling .pio\build\genericSTM32F411CE\src\main.cpp.o
> Linking .pio\build\genericSTM32F411CE\firmware.elf
> c:/users/test/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/12.3.1/../../../../arm-none-eabi/bin/ld.exe: warning: .pio/build/genericSTM32F411CE/firmware.elf has a LOAD segment with RWX permissions
> Checking size .pio\build\genericSTM32F411CE\firmware.elf
> Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
> RAM:   [          ]   1.2% (used 1604 bytes from 131072 bytes)
> Flash: [=         ]   7.9% (used 41468 bytes from 524288 bytes)
> Building .pio\build\genericSTM32F411CE\firmware.bin

To get rid of it, I had to use the search for a solution.
There were several mentions [(here](https://stackoverflow.com/questions/73429929/gnu-linker-elf-has-a-load-segment-with-rwx-permissions-embedded-arm-project) and [here)](https://wiki.st.com/stm32mcu/wiki/STM32CubeIDE:STM32CubeIDE_errata_1.15.x#General_issues) of similar problems, but no clear solution was found.
Through some experiments, I managed to eliminate the warning during compilation, the solution is below:
1. It is necessary to find the linker file of the microcontroller for which the firmware is being compiled, It is necessary to find the linker file of the microcontroller for which the firmware is compiled, for my MCU `stm32f411ceu6`  the path is as follows

> .platformio\packages\framework-arduinoststm32\variants\STM32F4xx\F411C(C-E)(U-Y)\ldscript.ld
2. Change the lines like this
```
  .preinit_array     :
  {
    . = ALIGN(4);
    PROVIDE_HIDDEN (__preinit_array_start = .);
    KEEP (*(.preinit_array*))
    PROVIDE_HIDDEN (__preinit_array_end = .);
    . = ALIGN(4);
  } >FLASH

  .init_array (READONLY) :
  {
    . = ALIGN(4);
    PROVIDE_HIDDEN (__init_array_start = .);
    KEEP (*(SORT(.init_array.*)))
    KEEP (*(.init_array*))
    PROVIDE_HIDDEN (__init_array_end = .);
    . = ALIGN(4);
  } >FLASH

  .fini_array (READONLY) :
  {
    . = ALIGN(4);
    PROVIDE_HIDDEN (__fini_array_start = .);
    KEEP (*(SORT(.fini_array.*)))
    KEEP (*(.fini_array*))
    PROVIDE_HIDDEN (__fini_array_end = .);
    . = ALIGN(4);
  } >FLASH
```
Add `(READONLY)` for two functions except .preinit_array, although the example above in the link said the opposite in the `STM32CubeIDE errata 1.15.x` document.
After this, the firmware will be installed without warnings.

brightproject, 2024-08-11:
Update:
In the [framework](https://registry.platformio.org/tools/platformio/framework-arduinoststm32) itself, this was somehow solved by prohibiting the output of flags, but this is not a good solution.
https://github.com/stm32duino/Arduino_Core_STM32/blob/5e4f22065273339bbd74c01b8209e9592d2c30bd/platform.txt#L82
The same linker has an error
https://github.com/stm32duino/Arduino_Core_STM32/blob/5e4f22065273339bbd74c01b8209e9592d2c30bd/variants/STM32F4xx/F411C(C-E)(U-Y)/ldscript.ld#L103

fpistm, 2024-08-12:
Hi @brightproject 
Here we support only Arduino IDE not PIO.
I know we simply disable the Warning because it is linked to the Arm none eabi gcc version used.
If the linker script used is updated with `READONLY` keyword then user using CMake with older version of the toolchain simply can't build. So what is the good solution?

fpistm, 2024-08-12:
Just generating some linker script with STM32CubeMX.
Now it uses `READONLY` with a warning, maybe we could do like that:
```
  .ARM.extab (READONLY) : /* The "READONLY" keyword is only supported in GCC11 and later, remove it if using GCC10 or earlier. */
  {
    . = ALIGN(4);
    *(.ARM.extab* .gnu.linkonce.armextab.*)
    . = ALIGN(4);
  } >FLASH

  .ARM (READONLY) : /* The "READONLY" keyword is only supported in GCC11 and later, remove it if using GCC10 or earlier. */
  {
    . = ALIGN(4);
    __exidx_start = .;
    *(.ARM.exidx*)
    __exidx_end = .;
    . = ALIGN(4);
  } >FLASH

  .preinit_array (READONLY) : /* The "READONLY" keyword is only supported in GCC11 and later, remove it if using GCC10 or earlier. */
  {
    . = ALIGN(4);
    PROVIDE_HIDDEN (__preinit_array_start = .);
    KEEP (*(.preinit_array*))
    PROVIDE_HIDDEN (__preinit_array_end = .);
    . = ALIGN(4);
  } >FLASH

  .init_array (READONLY) : /* The "READONLY" keyword is only supported in GCC11 and later, remove it if using GCC10 or earlier. */
  {
    . = ALIGN(4);
    PROVIDE_HIDDEN (__init_array_start = .);
    KEEP (*(SORT(.init_array.*)))
    KEEP (*(.init_array*))
    PROVIDE_HIDDEN (__init_array_end = .);
    . = ALIGN(4);
  } >FLASH

  .fini_array (READONLY) : /* The "READONLY" keyword is only supported in GCC11 and later, remove it if using GCC10 or earlier. */
  {
    . = ALIGN(4);
    PROVIDE_HIDDEN (__fini_array_start = .);
    KEEP (*(SORT(.fini_array.*)))
    KEEP (*(.fini_array*))
    PROVIDE_HIDDEN (__fini_array_end = .);
    . = ALIGN(4);
  } >FLASH
```

brightproject, 2024-08-12:
> If the linker script used is updated with `READONLY` keyword then user using CMake with older version of the toolchain simply can't build. So what is the good solution?

In any case, working code is better than compilation with warnings - you always have to find a balance.
My projects are 60% on `Arduino_Core_STM32,` and the rest are on `platformio`.
In the future, old platforms will go away anyway and new ones will take their place.
For now, of course, we need to support both old and new compilers.
Thank you for your feedback.

fpistm, 2024-08-21:
Hi @brightproject 
I've made a PR  to fix this. Anyway see my comment:
https://github.com/stm32duino/Arduino_Core_STM32/pull/2490#issuecomment-2301603605

brightproject, 2024-08-22:
> Hi @brightproject I've made a PR to fix this. Anyway see my comment: [#2490 (comment)](https://github.com/stm32duino/Arduino_Core_STM32/pull/2490#issuecomment-2301603605)
> 
> For ref: https://www.redhat.com/en/blog/linkers-warnings-about-executable-stacks-and-segments

Thank you very much for the correction and clarification - this is a new topic for me, I read it with pleasure🙂

PR #2490 (2024-09-04): chore: mark some segments as READONLY
This PR reverts 794945c10abb2dc6a875c471bb9aa4cbb131e5a2 then update all linker scripts to mark some segments as `READONLY`.    

Prevent LOAD segment with RWX permissions warning.
The `READONLY` keyword is only supported in GCC11 and later, remove it if using GCC10 or earlier.

Fixes #2475.
commit 05fbc11a (2024-09-04): chore: mark some segments as READONLY

Prevent LOAD segment with RWX permissions warning.
The "READONLY" keyword is only supported in GCC11 and later,
remove it if using GCC10 or earlier.

Fixes #2475.
closed 2024-09-04 by fpistm