stm32duino/Arduino_Core_STM32 #2477: Support for STM32H503CB

fpistm, 2024-08-12:
Could support for STM32H503CB be added? WeActStudio has released a pretty nice [board ](https://github.com/WeActStudio/WeActStudio.STM32H503CoreBoard)using that chip and probably there are more to come with H5 MCUs.

_Originally posted by @F14V in https://github.com/stm32duino/Arduino_Core_STM32/issues/1978#issuecomment-2273148024_

F14V, 2024-08-12:
Thank you for opening a proper issue from my comment and also for the work!

fpistm, 2024-08-13:
> Thank you for opening a proper issue from my comment and especially for the work!

Welcome, anyway avoid comment on closed issue. :wink:

justinjools22, 2024-09-22:
Hi, I was just looking for WeAct STM32H503CBU6 Arduino support. 
Does board part Generic STM32H503KBUx work? If not do you when support will be added for this board?