stm32duino/Arduino_Core_STM32 #2491: [New Variant] Generic STM32U073

doctek, 2024-08-21:
**Board description (please complete the following information):**
 - Name: Generic STM32U073KB
 - Link to all relevant information: Reference is to the STM data for the part. Nothing special here.
   * Documentations
   * Schematics
   * Where this hardware can be purchased
  * ...

**Note that user can add a STM32 based board following this [wiki](https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-(board))**

Any contribution is welcome, so do not hesitate to submit a PR.

All new variant request will be moved to [New Variant] request list #722 and the related issue will be closed.

fpistm, 2024-08-21:
Hi @doctek 
Before adding the support of the generic STM32U073, the STM32U0xx series have to be added.

doctek, 2024-08-21:
Thank you for your improvements to my request, fpistm! I appreciate it greatly. 
Any idea on a time line for this?
Thanks!

fpistm, 2024-08-22:
Hi @doctek 
I've made a first part of the job this morning using a script dedicated to add new stm32 series.
I've got the Blink working on a Nucleo U083RC.
I will do the PR this afternoon for basic support feel free to test on your side and give your feedback.

doctek, 2024-08-22:
This is amazing! I can' imagine how you were able to do this so quickly!! 
Of course I will gladly test your work on my STM32U073KB. I've read the instructions on Using the GIT Repository and I'm pretty confident I can follow that. The only thing I'm not sure of is what version to clone. What is the new version number?
Thank you for your hard work and rapid response!

fpistm, 2024-08-22:
You can fetch the PR or use my branch on my fork.

doctek, 2024-08-23:
Obviously, I'm not the GIT pro that you are, so let me make my question very clear.
In the wiki entry for "Using git repository", Step 3 says I should clone the new version using this command:
git clone https://github.com/stm32duino/Arduino_Core_STM32.git <version>
What should I put in for <version>?

Thanks, sorry for being stupid.

           -jim

doctek, 2024-08-24:
Hmm. That last posting didn't come through very clearly. I'm very sorry for that, although I don't know exactly what caused it.
The wiki reference should have been git clone [https://] [github.com/] [stm32duino/Arduino_Core_STM32.git] [<[version]>]
Hopefully, I've broken the command up enough that it will post correctly.
My question is:  What should I put in for [version]?

Thanks and sorry for messing up the last post and not noticing it!
           -jim

fpistm, 2024-08-24:
Hi @doctek 
Sorry was OoO yesterday. Version is the cire version. So 2.8.1.

doctek, 2024-08-24:
Following the instructions, I deleted the 2.8.1 files that I had and cloned the latest version. 
When I started Arduino, I do not see any STM board choice that is for the U073. What am I missing?
Looking at the git files for 2.8.1, I don't see any reference to merging #2495. Makes ;me think that I am not cloning the correct version. Again, I'm not a git power user! Can you tell me what I'm doing wrong?
Thanks for your patience!
     -jim

fpistm, 2024-08-26:
PR #2495 is not merged.
As stated you can:
- Clone my fork and checkout my branch:
https://github.com/fpistm/Arduino_Core_STM32/tree/STM32CubeU0_addition

- Fetch only the PR in the original repo. For this I have a git alias in my `.gitconfig`:
```
[alias]
    pr = "!f() { git fetch -fu ${2-origin} refs/pull/$1/head:pr/$1 && git checkout pr/$1; }; f"
```
when added you can simply do on the git repo:
`git pr 2495`

doctek, 2024-08-27:
I'm trying real hard to follow your instructions.
Here's what I did. You'll have a good laugh at my git newbee  errors!
First, I tried to clone your repository. I deleted 2.8.1 and issued this command:
git clone https://github.com/fpistm/Arduino_Core_STM32
I tried to clone the complete link you gave https://github.com/fpistm/Arduino_Core_STM32/tree/STM32CubeU0_addition, but that wouldn't work.
Then I modified the file .git/config to add the [alias] for pr.
Then I issued this command:
git pr 2495
Result:
fatal: couldn't find remote ref refs/pull/2495/head

I have no idea what to try next. Thank you for your patience.

doctek, 2024-08-27:
Actually, I did 
git clone https://github.com/fpistm/Arduino_Core_STM32 2,8,1
The rest is correct.

doctek, 2024-08-27:
Progress! This command did it.
git clone --single-branch --branch STM32CubeU0_addition https://github.com/fpistm/Arduino_Core_STM32.git 2.8.1
I fired up Arduino and there is now a generic STM32U0 board type!
Of course, it's never quite that simple. I'm using an STM32U073KBU, not any of the STM32U73R versions listed. I *think* I can bend one of these enough to let me test the code. I'll keep you posted.
Again, many thanks for your support!

doctek, 2024-08-27:
OK, I did some more poking around in variants directory and found the U073KBU variant! I think a little work on the boards file will get this working for now.

fpistm, 2024-08-27:
git pr alias is to fetch a PR on the official repo. So you have to clone the official repo not my fork. Anyway cloning my fork do the same and it is easy to fetch update.

doctek, 2024-08-28:
Happy Dance time! My STM32U073KBU board blinks a led!
Here's what I had to do:
- must use arduino-cli. The Arduino ide doesn't see any 073 boards except the R versions. Efforts to add names to the boards.txt file didn't work.
-  must add ldscript.ld to the variants file for the U073 boards.
- must update STM32CubeProgrammer to the latest version to get the loader.
With that done, it works! Can't wait for it to be properly integrated into Arduino, but I can keep working.
Thank you for your excellent work!
            -jim

fpistm, 2024-08-29:
> * must use arduino-cli. The Arduino ide doesn't see any 073 boards except the R versions. Efforts to add names to the boards.txt file didn't work.

You probably need to clean up Arduino IDE cache.

doctek, 2024-08-29:
Clearing the cache is a good suggestion. Googling for how to do that offered only one useful suggestion: remove ard* files from /tmp. So I tried that, but the new boards still do not appear, although I have added them to the boards.txt file. Obviously, there are  additional files cached by Arduino, since it knew what sketch I was editing even after I deleted the ard* files and restarted Arduino. 
Could you be so kind as to tell me how to purge the Arduino cache and make the board types appear?
Thank you!

doctek, 2024-08-30:
I found more about clearing the Arduino cache. Using the sledge hammer approach of deleting the ~/.config/arduino-ide file and restarting did the trick. Now I see the U073K entries. I expect they will work as desired.
Is there a gentler way to clear the cache?
Thanks!

fpistm, 2024-08-30:
This is the method I used. An issue is opened on Arduino IDE to fix this issue.

doctek, 2024-08-30:
That did it! I can program mySTMU073K parts from the Arduino IDE.
Thanks again for the excellent support and patient assistance.
           -jim

fpistm, 2024-09-03:
For ref:
https://github.com/arduino/arduino-ide/issues/1030#issuecomment-1152005617

PR #2495 (2024-09-05): feat: add STM32U0xx support
This PR add the support of the [STM32U0](https://www.st.com/en/microcontrollers-microprocessors/stm32u0-series.html) series.

* From [STM32CubeU0 v1.1.0](https://github.com/STMicroelectronics/STM32CubeU0/releases/tag/v1.1.0)
  * Add HAL from version 1.1.0
  * Add CMSIS from version 1.0.0

## Tests using a [Nucleo-U083RC](https://www.st.com/en/evaluation-tools/nucleo-u083rc.html):
- [x] GPIO
- [x] ADC
- [x]  ADC internal (see https://github.com/stm32duino/STM32Examples/pull/59)
- [x] HardwareTimer (PWM)
- [x] EXTI
- [x] I2C (EEPROM RW, I2C scanner)
- [x] SPI (SD, TFT [... cut]
closed 2024-09-05 by fpistm