stm32duino/Arduino_Core_STM32 #2511: 2.8 causes ALL "Ports" in Tools Menu to disappear...

WestfW, 2024-09-09:
This is essentially a duplicate of [@2469](https://github.com/stm32duino/Arduino_Core_STM32/issues/2469)
Except I'm clarifying that installing 2.8.x STM board package in a 1.8.x Arduino causes the Ports menu item to be grayed out in the tools menu FOR ALL BOARDS, not just when an ST board is selected.
Granted that ST Board Package 2.8.x is not supported except in Arduino IDE 2.x, it is still unacceptable that installing it would break all other platforms.
(I don't even understand how that's possible.  Isn't the "Ports" menu computed entirely by the IDE, without involving platform code?)

fpistm, 2024-09-09:
As stated by you it is a duplicate of #2469.
The fact the port disappear from all other core is an Arduino issue, this is the IDE which is in charge of the COM port display.
Can't do nothing here and will not investigate it. Arduino IDE 1.8.x support is dropped since 2.8.x and is clearly warned.

WestfW, 2024-09-11:
I did find a fix for this, and it's something that might cause a problem even with the 2.x IDE...

Running the IDE in a terminal window, I get likely looking error messages:

```txt
Exception in thread "cc.arduino.packages.discoverers.serial.SerialDiscovery" java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
        at java.util.LinkedList.checkElementIndex(LinkedList.java:555)
        at java.util.LinkedList.get(LinkedList.java:476)
        at processing.app.Platform.resolveDeviceByVendorIdProductId(Platform.java:189)
        at cc.arduino.packages.discoverers.serial.SerialDiscovery.forceRefresh(SerialDiscovery.java:166)
        at cc.arduino.packages.discoverers.serial.SerialDiscovery$1.run(SerialDiscovery.java:96)
        at java.util.TimerThread.mainLoop(Timer.java:555)
```

The major difference between ST's board package in 2.7.x and 2.8.x that has anything to do with SerialDiscovery is that 2.8.x added MANY vid/pid combinations to their boards.txt, using the xxx.vid.7 xxx.pid.7 format.

You might think from the error message that they managed to simply overflow some array inside the Arduino IDE, but it also implies linked lists, that ought to be able to grow indefinately...

And the actual problem apparently turns out to be that ONE of the vendors ("Blues"?) boards.txt entry was done incorrectly.  I think it managed to add a xxx.pid.0 entry without a matching xxx.vid.0 entry, so that when the discovery tried to get a pair, the vid was simply not found.

I don't know if this would also cause problems with the Blues boards in IDE 2.0.  It seems like it could...

Anyway, the diffs in the boards.txt file that fix the problem look like this:

```diffs
--- ../Java/portable/packages/STMicroelectronics/hardware/stm32/2.8.1/boards.txt	2024-07-18 15:00:00.000000000 -0700
+++ /tmp/boards.txt.new1	2024-09-11 01:05:41.000000000 -0700
@@ -12304,7 +12304,6 @@
 Blues.build.flash_offset=0x0
 Blues.upload.maximum_size=0
 Blues.upload.maximum_data_size=0
-Blues.vid.0=0x30A4
 
 # Swan R5 board
 Blues.menu.pnum.SWAN_R5=Swan R5
@@ -12318,6 +12317,7 @@
 Blues.menu.pnum.SWAN_R5.build.product_line=STM32L4R5xx
 Blues.menu.pnum.SWAN_R5.build.variant=STM32L4xx/L4R5Z(G-I)Y_L4R9Z(G-I)Y_L4S5ZIY_L4S9ZIY
 Blues.menu.pnum.SWAN_R5.build.peripheral_pins=-DCUSTOM_PERIPHERAL_PINS
+Blues.menu.pnum.SWAN_R5.vid.0=0x30A4
 Blues.menu.pnum.SWAN_R5.pid.0=0x0002
 Blues.menu.pnum.SWAN_R5.debug.server.openocd.scripts.2=target/stm32l4x.cfg
 Blues.menu.pnum.SWAN_R5.debug.svd_file={runtime.tools.STM32_SVD.path}/svd/STM32L4xx/STM32L4R5.svd
@@ -12334,7 +12334,8 @@
 Blues.menu.pnum.CYGNET.build.product_line=STM32L433xx
 Blues.menu.pnum.CYGNET.build.variant=STM32L4xx/L433C(B-C)(T-U)_L443CC(T-U)
 Blues.menu.pnum.CYGNET.build.peripheral_pins=-DCUSTOM_PERIPHERAL_PINS
-Blues.menu.pnum.CYGNET.build.pid=0x0003
+Blues.menu.pnum.CYGNET.build.vid.0=0x30A4
+Blues.menu.pnum.CYGNET.build.pid.0=0x0003
 Blues.menu.pnum.CYGNET.debug.server.openocd.scripts.2=target/stm32l4x.cfg
 Blues.menu.pnum.CYGNET.debug.svd_file={runtime.tools.STM32_SVD.path}/svd/STM32L4xx/STM32L4x3.svd
 
```

fpistm, 2024-09-11:
Fix should be 
```
-Blues.menu.pnum.CYGNET.build.pid=0x0003
+Blues.menu.pnum.CYGNET.build.pid.0=0x0003
```
vid.0 definition is correct and global to the Blues menu entry.

closed 2024-09-11 by fpistm

WestfW, 2024-09-12:
Thanks!  I thought I tried just fixing the per-board pids, and it didn't work.  But it might have been a different permutation...

WestfW, 2024-09-12:
(I checked.  The simpler change doesn't fix the "ports" menu disappearing in 1.8.x)

fpistm, 2024-09-12:
> (I checked. The simpler change doesn't fix the "ports" menu disappearing in 1.8.x)

As stated it is no more compatible with 1.8.x and so no more support for it.

WestfW, 2024-09-13:
I'll point out:

1. The more complex patch allows the STMDuino core to work on BOTH 1.8.x and 2.x
2. The "Blues" boards are the only entries in the board.txt file that attempt to use this "inheritance" of .vid.0  (are you sure this works, even in 2.x Arduino IDE?)

fpistm, 2024-09-13:
Your patch is not correct as you defined "build.vid" which is not expected.

Could you try this:

```patch
--- ../Java/portable/packages/STMicroelectronics/hardware/stm32/2.8.1/boards.txt	2024-07-18 15:00:00.000000000 -0700
+++ /tmp/boards.txt.new1	2024-09-11 01:05:41.000000000 -0700
@@ -12304,7 +12304,6 @@
 Blues.build.flash_offset=0x0
 Blues.upload.maximum_size=0
 Blues.upload.maximum_data_size=0
-Blues.vid.0=0x30A4
 
 # Swan R5 board
 Blues.menu.pnum.SWAN_R5=Swan R5
@@ -12318,6 +12317,7 @@
 Blues.menu.pnum.SWAN_R5.build.product_line=STM32L4R5xx
 Blues.menu.pnum.SWAN_R5.build.variant=STM32L4xx/L4R5Z(G-I)Y_L4R9Z(G-I)Y_L4S5ZIY_L4S9ZIY
 Blues.menu.pnum.SWAN_R5.build.peripheral_pins=-DCUSTOM_PERIPHERAL_PINS
+Blues.menu.pnum.SWAN_R5.vid.0=0x30A4
 Blues.menu.pnum.SWAN_R5.pid.0=0x0002
 Blues.menu.pnum.SWAN_R5.debug.server.openocd.scripts.2=target/stm32l4x.cfg
 Blues.menu.pnum.SWAN_R5.debug.svd_file={runtime.tools.STM32_SVD.path}/svd/STM32L4xx/STM32L4R5.svd
@@ -12334,7 +12334,8 @@
 Blues.menu.pnum.CYGNET.build.product_line=STM32L433xx
 Blues.menu.pnum.CYGNET.build.variant=STM32L4xx/L433C(B-C)(T-U)_L443CC(T-U)
 Blues.menu.pnum.CYGNET.build.peripheral_pins=-DCUSTOM_PERIPHERAL_PINS
-Blues.menu.pnum.CYGNET.build.pid=0x0003
+Blues.menu.pnum.CYGNET.vid.0=0x30A4
+Blues.menu.pnum.CYGNET.pid.0=0x0003
 Blues.menu.pnum.CYGNET.debug.server.openocd.scripts.2=target/stm32l4x.cfg
 Blues.menu.pnum.CYGNET.debug.svd_file={runtime.tools.STM32_SVD.path}/svd/STM32L4xx/STM32L4x3.svd
 
```

> 2\. The "Blues" boards are the only entries in the board.txt file that attempt to use this "inheritance" of .vid.0  (are you sure this works, even in 2.x Arduino IDE?)

Yes, it works with IDE 2.x like several other recipes which can be redefined at board level.

WestfW, 2024-09-14:
Yes, that patch sees to work fine.

Analysis by the Arduino folks is here:
https://forum.arduino.cc/t/arduino-ide-1-8-19-in-tools-menu-port-grayed-out/1287639/23

TLDR: IDE 2.x isn't affected, and they like your patch  as well.

Thank you: this was more attention and faster response than I expected for support of obsolete tools, and from your initial responses.

PR #2512 (2024-09-11): fix(boards.txt): use pid.0 instead of pid for Blues Cygnet
to be aligned with vid.0.

Fixes #2511.
commit 89a29516 (2024-09-11): fix(boards.txt): use pid.0 instead of pid for Blues Cygnet

to be aligned with vid.0.

Fixes #2511.