stm32duino/Arduino_Core_STM32 #2514: spi transfer16 is not 16 bit, and excessive overhead

drmcnelson, 2024-09-12:
**Describe the bug**

SPI transfer16()   produces two 8 byte transfers rather than one 16 bit transfer.

**To Reproduce**
Here is a code that is used for testing SPI transfers.   In this instance, the CS pin tells an ADC to convert its input to digital, then 730 nanoseconds later the data is ready for a 16 bit SPI transfer.  

With an SPI clock of 50kHz,  this should be able to run at close to 1MSPS.

       for (int n = 0; n < NKNTS; n++ ) {
 
       // For the ADC
	DIGITALWRITE(CSPin,HIGH);
	DELAYNANOSECONDS(ADC_WAIT_NANOSECONDS);
	DIGITALWRITE(CSPin,LOW);
	
	// For the oscilloscope
	DIGITALWRITE(SPAREPIN,HIGH);  

        // 16 bit transfer
	data[n] = SPI.transfer16(0xFFFF);

	// For the oscilloscope       
	DIGITALWRITE(SPAREPIN,LOW);

      }
`

And here is the  oscilloscope trace.   Notice that rather than one 16 bit transfer, we have two 8 bit transfers.  And they are separated by 500 nsec.

[image]

Here is the implementation, for transfer16() it calls  spi_transfer( .... length )

https://github.com/stm32duino/Arduino_Core_STM32/blob/89a29516cca7b50c6aef17a52e0e6e4afe78aeb7/libraries/SPI/src/SPI.cpp#L179-L195

And, in utility/spi.c,  we find that spi_transfer() simply loops over  LL_SPI_TransmitData8( ).

https://github.com/stm32duino/Arduino_Core_STM32/blob/89a29516cca7b50c6aef17a52e0e6e4afe78aeb7/libraries/SPI/src/utility/spi_com.c#L482-L550

You already have a  LL_SPI_TransmitData16( ),  under hardware/stm32/2.8.1/system/Drivers/

What is missing in this implementation is a spi_transfer16() and to change transfer16() to call it.

Thank you

fpistm, 2024-09-13:
Hi @drmcnelson 
This is usual implementation in several Arduino core:
https://github.com/arduino/ArduinoCore-renesas/blob/fedf7896ed6c2e66d42ed5642198f5617a2daa55/libraries/SPI/SPI.cpp#L206-L219
https://github.com/arduino/ArduinoCore-sam/blob/790ff2c852bf159787a9966bddee4d9f55352d15/libraries/SPI/src/SPI.cpp#L199-L214
https://github.com/arduino/ArduinoCore-megaavr/blob/5e639ee40afa693354d3d056ba7fb795a8948c11/libraries/SPI/src/SPI.cpp#L238-L249
https://github.com/arduino/ArduinoCore-samd/blob/993398cb7a23a4e0f821a73501ae98053773165b/libraries/SPI/SPI.cpp#L242-L253

It could probably be enhance.
Feel free to provide a PR.

drmcnelson, 2024-09-13:
Yes, notice that they are derived from the same earlier version, The earliest version that I found, ran on the uno, an 8bit 16MHZ legacy platform marketed to hobbyists.

Adapting it in form to the much faster 16 and 32 bit platforms, leaving the internal kluges and inefficiencies in place was an amateur kluge.     And it is simply not correct.  

There are two errors,  (a) 16 bit transfers implemented as 8 bit transfers, and (b) failure to also provide a loop friendly interface (i.e., separate calls for setup and transfer, rather than forcing setup and transfer into every call)

The erroneous implementation of 16 bit transfers as two 8 bit transfers, alone, reduces performance by something like a factor of 2 or more.   Forcing setups into very transfer and not providing a loop friendly API,  reduces performance by a solid factor of 8. 

The result is that the API is therefore incompatible with a large class of important peripherals; c.f., any device that requires a pin assertion between transfers and better than 250KSPS transfers.  An ADC reading a linear CCD would be a good example.  

Sometimes there has been push back on fixing this, it is usually about the transaction interface, and it usually comes from that same hobbyist crowd.   The logic does not hold up.   Reserving the device, even pushing the settings onto a stack, if you think you must despite it being an on-the-metal platform, should not be effected by what you do with it once it is reserved.

Concluding this introduction, lets note that  limiting a 200MHz MCU with a 50MHZ SPI clock to 250ksps transfers, only because the code was inherited from a slow 8 bit legacy platform that catered to hobbyists, does not make a lot of sense and it is not good for the platform.

 [My solution for that for the UNO R4, is here ](https://github.com/drmcnelson/Arduino_UNO_R4_SPI_Speedup).  There is also a solution for the Teensy 4.x, [the discussion is here](https://forum.pjrc.com/index.php?threads/spi-speedup-question.72003/)

In both of those, the patch (a) implements a true 16 bit transfer, and (b) provides two new calls, one  for the setup and another for the actual transfer.

BTW  The original title referred also to overheads.   That is still part of the problem.

A loop in a user program can look like this:

      SPI.transfer16_setup();
     for (n=0;n<2048;n++) {

         // Toggle the CONVERT for the ADC
         digitalWrite(some_pin,HI);
         nanosecondwait(500);
         digitalWrite(some_pin,LOW);

         // 200 nanosec to the first clock, so 700 nanosec total, as required by the ADC
         SPI.transfer16_transfer(inbuffer[n].outbuffer[n]);
     }

Note that with a 50MHZ SPI clock, this could run at close to 1MSPS.

The STM32 could be a serious platform.   It deserves a little more attention to the API for the SPI, one of its most important peripheral interfaces.

drmcnelson, 2024-09-13:
P/S, i forgot, in the loop-friendly API,  you may want a "restore" as well.   Looking at the transfer times. I see there is only 400 nsecs between the two 8 bit transfers,  but 1200nsecs before and 1100nsecs after.   So that means a correct loop-friendly call, besides the 16bits/50MHZ, only needs an additional 400nsecs.

fpistm, 2024-09-13:
Main issue with your setup/transfer api's is it is not Arduino api compatible.

drmcnelson, 2024-09-13:
What makes it incompatible?

a)  Those calls that I suggest, are what is  _missing_ from the Arduino API to make it actually useful.

b)  You already added new calls and functionality to the API,  setSSEL() etc.

drmcnelson, 2024-09-13:
Looking at the code again,  if you could save me a little time and explain what the conditional compiles are about in
https://github.com/stm32duino/Arduino_Core_STM32/commit/89a29516cca7b50c6aef17a52e0e6e4afe78aeb7
I might try a patch and you can see if you like it enough for a PR.

fpistm, 2024-09-16:
> What makes it incompatible?
> 
> a) Those calls that I suggest, are what is _missing_ from the Arduino API to make it actually useful.
> 
> b) You already added new calls and functionality to the API, setSSEL() etc.

Well several examples already exists, if new one are created using those new API they will make it incompatible with other core.
the `setxxx` are dedicated but only for setup purpose and it is easy to use it and having a sketch compatible with other core:
```
#if defined ARDUINO_ARCH_STM32
SPI.setxxx();
#endif
```
While new API simply change the way you implement the SPI transaction.

About the "_conditional compiles_", it is simply to have the code working for each STM32 series which can have different SPI IP and so required different configuration or API.

fpistm, 2024-09-16:
My concern is about the example (sketch) which "should" as much as possible be compatible between core. This is the way Arduino works.  That's all 😉 

> As long as you insist on this very incorrect position, your SPI implementation will be limited to taking 5 usecs for transfers that should take 320 nsecs.

I don't insist 😉 we discuss. I never told it will not be implemented. This is a community project, any contribution are welcome.

> Aside, your SPI.setxxx() would of course be backward compatible. And I am very interestred in setSSEL(), but unfortunately when I add that call to my setup(), it crashes.

See my answer on the forum -> https://www.stm32duino.com/viewtopic.php?p=14825#p14825
As usual wrong usage.

> The 50MHz SPI is a big deal, it is a terrible shame that is squandered by stopping short when you already did so much work and you are already so close.

Why? First goal was to provide support for STM32 series in Arduino with Arduino API references. I know it provides limitations and it could be improved but as always time is the missing ingredient. Several other features required to be implemented or enhanced., do my best. As stated any contribution are welcome.

drmcnelson, 2024-09-16:
> My concern is about the example (sketch) which "should" as much as possible be compatible between core. This is the way Arduino works. That's all 😉
> 

Correcting transfer16() to do 16bit transfers, and adding loop friendly calls should not change the way the existing example works.

> 
> I don't insist 😉 we discuss. I never told it will not be implemented. This is a community project, any contribution are welcome.
> 
Yes I apologize for sounding argumentative.   There is a lot of resistance and appeal to dogma in the community.  The transaction api is a good example. 

> See my answer on the forum -> https://www.stm32duino.com/viewtopic.php?p=14825#p14825 As usual wrong usage.
> 
Thank you, I am in the lab at the moment and dont have that log-in.  I see that it was the wrong pin., no doubt a result of failing to digest the datasheet.  

Are D13, D12, D11 then not the pins that it is using for the SPI?   My intent was to use the next pin, as is customary,

Also I wasnt sure which SPI it is using, it seems like SP1, 

> > The 50MHz SPI is a big deal, it is a terrible shame that is squandered by stopping short when you already did so much work and you are already so close.
> 
> Why? 

Because that it is what enables 1MSPS reads and that ties into a large swath of applications that are important in using arduino class boards for doing science.

> First goal was to provide support for STM32 series in Arduino with Arduino API references. I know it provides limitations and it could be improved but as always time is the missing ingredient. [...]

My philosophy is that while I  might not implement everything, I try to at least, not introduce any limitation not present in the hardware.   So there is a difference between fully supporting a rich set of features provided by the hardware, and for that I agree, who has time.   Unnecessarily choking off performance is something else, and that is what the historic SPI library does.   Fortunately, it can be fixed with a patch or two plus a few extra calls that do not disturb the API already in place.

drmcnelson, 2024-09-16:
Are you using CLA for PR's?

If not, then I might take a look at patching the library, perhaps even today.

It looks like you did a good job with factoring, so it seems like it should be easy.

fpistm, 2024-09-16:
> Are D13, D12, D11 then not the pins that it is using for the SPI? My intent was to use the next pin, as is customary,
> 
> Also I wasnt sure which SPI it is using, it seems like SP1,

D11 to D13 are usually the default SPI pins (for boards with Arduino UNO connector) and D10 used a CS. But in a general way CS is managed by the application (software). When you used `setSSEL()`, you ask to use the Hardware CS and so no need to drive the pin at application level.
For the Nucleo F722, unfortunately the D10 pin  does not have this capability. 

To go deeper, when a board  have a Arduino connector, default SPI pins are D10 to D13:
https://github.com/stm32duino/Arduino_Core_STM32/blob/89a29516cca7b50c6aef17a52e0e6e4afe78aeb7/cores/arduino/pins_arduino.h#L43-L65
In our case for Nucleo F722:
https://github.com/stm32duino/Arduino_Core_STM32/blob/89a29516cca7b50c6aef17a52e0e6e4afe78aeb7/variants/STM32F7xx/F722Z(C-E)T_F732ZET/variant_NUCLEO_F722ZE.h#L29-L31
Thanks to the `PeripheralPins.c`:
https://github.com/stm32duino/Arduino_Core_STM32/blob/89a29516cca7b50c6aef17a52e0e6e4afe78aeb7/variants/STM32F7xx/F722Z(C-E)T_F732ZET/PeripheralPins.c#L288
https://github.com/stm32duino/Arduino_Core_STM32/blob/89a29516cca7b50c6aef17a52e0e6e4afe78aeb7/variants/STM32F7xx/F722Z(C-E)T_F732ZET/PeripheralPins.c#L307
https://github.com/stm32duino/Arduino_Core_STM32/blob/89a29516cca7b50c6aef17a52e0e6e4afe78aeb7/variants/STM32F7xx/F722Z(C-E)T_F732ZET/PeripheralPins.c#L322

So to answer your question SPI uses `SPI1`.

Finally, you can check in the below table which SPI SSEL to use:
https://github.com/stm32duino/Arduino_Core_STM32/blob/89a29516cca7b50c6aef17a52e0e6e4afe78aeb7/variants/STM32F7xx/F722Z(C-E)T_F732ZET/PeripheralPins.c#L338-L349

> Are you using CLA for PR's?

No. Only follow the [CONTRIBUTING.md](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/CONTRIBUTING.md)

drmcnelson, 2024-09-16:
Thank you, quite elaborate and seems like several layers of naming.   So it seems like SPI1 uses PA_5,PA_6,PA_7 and PA_4 for CS.

What is confusing then, is this from the user manual (UM1974), page 58, where it says the arduino SPI uses PA5, PA6, PA7,  as D13,D12,D11,  and for D10 it is not PA4, but rather PD14

[image]

drmcnelson, 2024-09-16:
I found it, page 32 in the user manual.   If one uses an Arduino shield adapter, the pin is covered.  So that means it needs a custom adapter board.  Not a big deal, the first thing though is to see the timing and whether it works with the ADC.

[image]

fpistm, 2024-09-17:
> So, bottom line, what goes in the call to setSSEL(), and what pin does it appear on?

Well, if PD14 is described as SPI1_CS then there is a mistake in the datasheet or in the stm32 open pin data: https://github.com/STMicroelectronics/STM32_open_pin_data/blob/f4ec11f00e762e37ffc4020f6d4f20d225bc061d/mcu/STM32F722Z(C-E)Tx.xml#L626-L632

The PeripheralPins.c is generated from this file that's why PD14 is not in the SPI SSEL array. 
Looking at the STM32F722ZE datasheet, there is no SPI_CS for PD14:
[image]

My guess is the Nucleo F722ZE datasheet simply tell that PD14 can be used for CS on the Arduino connector to be compatible. As Arduino manages CS by software (gpio toggle) and not by hardware (manage by the SPI IP itself).

> I found it, page 32 in the user manual. If one uses an Arduino shield adapter, the pin is covered. So that means it needs a custom adapter board. Not a big deal, the first thing though is to see the timing and whether it works with the ADC.

About this it is not page 32 but 35 even if several pins match.
[image]

drmcnelson, 2024-09-18:
Okay,  so the code compiles and runs using PA_4 or D24.  i.e. SPI.setSSEL(PA_4) and SPI.setSSEL(24), both compile and run.

In the following, the traces from top to bottom are yellow = D13 (SCK), blue = D12 (MISO), red = D11 (MOSI), and green = PA_4 (D24).

Notice that the first time the SPI is called, the  MOSI is low and goes high for the transfers.   That is actually not correct,  It needs to be high when not active.

The second time the SPI loop is invoked, MOSI stays high.   So it is not consistent,.

The  PA_4 pin (SSE)L does nothing.

[image]

[image]

And here is the interface using D10 for the CSEL, under program control.  You can see the MOSI again starts low, and then goes to the high state after the second pair of 8 bit transfers.

[image]

drmcnelson, 2024-09-22:
Okie dokie,

Here is an implementation for 16 bit transfers.   The attached is a zip of the SPI/src tree with the 16 bit transfer implemented.  I left the old routine in place with a suffix on the name.   The mod for the 16 bit transfer is pretty simple, really.  Does it look right?

[SPI_src.zip](https://github.com/user-attachments/files/17091276/SPI_src.zip)

The good news is that there seems to be no per-transfer setup that could be optimized out of the call for loops.  So there is nothing that requires or would benefit from a loop friendly API.   (I am still testing this, so please consider it preliminary).

There are two items of less good news.

A)  The overheads anyway are large, and for the one before transfer starts it is not a constant delay.  So that is a big deal for some real time applications.  

[image]

I think the delays comes from here, does that seem right? 

https://github.com/stm32duino/Arduino_Core_STM32/blob/c888cf97867f04185c1e6120b867b2736229b645/libraries/SPI/src/utility/spi_com.c#L502-L507

https://github.com/stm32duino/Arduino_Core_STM32/blob/c888cf97867f04185c1e6120b867b2736229b645/libraries/SPI/src/utility/spi_com.c#L510-L514

Aside, for the ADC, there is a pin CNVST (connected to the SSEL line) that starts the conversion, conversion then takes 700nsecs and the CNVST pin needs to be cleared at least 30 nsecs before readout.    That means to get best performance,  I would have to hack that into the transfer routine, i.e. set it, do the idiot loop for the interface ready flag, then clear cnvst, and start the transfer.   Doable, but tedious.

B)  The MOSI line is, well,  flakey, and not completely correct.

I am guessing there is a setup or config somewhere that controls the idle state for the MOSI line, is that correct?

Here is the aberrant behavior that needs to be fixed:

  i)  The apparent idle state before the first call is low (should be high)

  ii)  For txdata, it goes high momentarily for the first transfer,   that is half-correct (see above, i)

 iii)  For the second transfer it goes high and stays high.  It stays high after this.

[image]

The correct  or at lease more conventional, behavior for txdata 0xFFFF, of course, is that it starts high and stays high.

fpistm, 2024-09-23:
> Does it look right?
> 
> [SPI_src.zip](https://github.com/user-attachments/files/17091276/SPI_src.zip)

Please create a PR in draft. Archive is really not efficient to track and comment.

drmcnelson, 2024-09-23:
The PR draft is submitted.

I also want to take up the issue with the MOSI pin state.   That too is critical.   Shall I make a new issue for it?

fpistm, 2024-09-23:
> I also want to take up the issue with the MOSI pin state. That too is critical. Shall I make a new issue for it?

No. It is not a bug. MOSI pins kept the last value level. We have no way to specify the level when CS is not active. And see no reference it should be high in SPI documentations. It is simply "don't care"