stm32duino/Arduino_Core_STM32 #2520: digital and analog read/write functions do not work with PinName

PerennialNovice, 2024-09-18:
The standard Arduino functions like digitalWrite(pin, state) do accept PinName pins (like PA_5, PF_11), but due to implicit conversion or integer promotion and conversion - not sure about the mechanism behind it - they turn out to be applied on a different pin, or no pin at all.

Adding a set of "wrapper" functions would do the trick here:

```
void pinMode(PinName pin, uint32_t mode)
{
	pinMode(pinNametoDigitalPin(pin), mode);
}
int digitalRead(PinName pin)
{
	return digitalRead(pinNametoDigitalPin(pin));
}
void digitalWrite(PinName pin, uint32_t val)
{
	digitalWrite(pinNametoDigitalPin(pin), val);
}
void digitalToggle(PinName pin)
{
	digitalToggle(pinNametoDigitalPin(pin));
}
uint32_t analogRead(PinName pin)
{
	return analogRead(pinNametoDigitalPin(pin));
}
void analogWrite(PinName pin, uint32_t val)
{
	analogWrite(pinNametoDigitalPin(pin), val);
}
```

though this seems to work in my use case, I am not sure that everything is covered by these simple functions ...

closed 2024-09-18 by fpistm

PerennialNovice, 2024-09-18:
Thanks for the explanation. I missed the C-function part because, as you wrote, I define them in the sketch...

As for the interest, we have a minimal advantage, because we store some pins in an EEPROM and using only the pin number, the alternate functions get dropped, which renders it impossible to deduct i.e. timer peripheral from the stored pin.
So we chose to store the PinNames which has all relevant information.

I nevertheless agree that this is a very special use case, and might be irrelevant to most - if not all - of the other users.

fpistm, 2024-09-18:
> using only the pin number, the alternate functions get dropped,

PinName does not include this kind of information you alway needs to get it from the dedicated array stored in the PeripheralPins.c.
So simply do the conversion using `digitalPinToPinName`