stm32duino/Arduino_Core_STM32 #2534: I2C Speed incorrect for I2C5 on Nucleo-H723ZG board

Steffen-Zienert, 2024-10-21:
**Describe the bug**
The speed for the I2C5 peripheral is always ~16.8kHz and not the expected 100kHz on the Nucleo-H723ZG board and possibly other H7 variants. It does not change for different values of `Wire.setClock(values)`. The other I2C peripherals behave as expected.

**To Reproduce**
Minimal reproducible example code:
```
#include <Wire.h>

void setup() {
  // set Wire to Pins associated with I2C5:
  Wire.setSDA(PC_10);
  Wire.setSCL(PC_11);
  Wire.begin();
  Wire.setClock(100000);

  // send any I2C message
  Wire.beginTransmission(0x24);
  Wire.write(0);
  Wire.endTransmission();
}

void loop() {
}
```
Results in ~16.8kHz on SCL Pin PC11 (see image)

**Expected behavior**
Expect 100kHz clock speed.

**Screenshots**
Measurement of Pin PC 10 (red) and PC11 (brown): 
[image]
Clock speed around 16.8kHz.

**Desktop (please complete the following information):**
 - OS: Windows 11 
 - Arduino IDE version: 2.3.3
 - STM32 core version: 2.81
 - Tools menu settings: default
 - Upload method: Mass storage

**Board (please complete the following information):**
 - Name: Nucleo-H723ZG
 - Hardware Revision: Rev E01
 - Extra hardware: no

**Additional context**
I think that the issue is in the `i2c_getClkFreq(I2C_TypeDef *i2c)` function. I single stepped the code and this snipped always returns 0 for the I2C5 peripheral clock frequency `clkSrcFreq`:
https://github.com/stm32duino/Arduino_Core_STM32/blob/42e02623a246f06fb5748b362c2087d04bb37c0e/libraries/Wire/src/utility/twi.c#L415-L424
For the working I2C1 Peripheral with the same clock source there is handling for that case:
https://github.com/stm32duino/Arduino_Core_STM32/blob/42e02623a246f06fb5748b362c2087d04bb37c0e/libraries/Wire/src/utility/twi.c#L207-L254

fpistm, 2024-10-21:
Hi @Steffen-Zienert 
Thanks for the detailed issue.
I've made a fix (#2535), could you test it, please ?

Steffen-Zienert, 2024-10-21:
Hi @fpistm,

thanks for the fast reply. My problem is fixed with your new code. The MRE behaves as expected:

[image]

PR #2535 (2024-10-21): fix(i2c): wrong clock frequencies for I2C5 and I2C6
Fixes #2534
commit 511ed73b (2024-10-21): fix(i2c): wrong clock frequencies for I2C5 and I2C6

Fixes #2534
closed 2024-10-21 by fpistm