stm32duino/Arduino_Core_STM32 #2551: __libc_init_array fails on STm32H750XBH6

thisdp, 2024-11-05:
Arduino IDE 2.3.3
STM32H750XBH6
STM32 MCU Based Boards 2.8.1
An empty sketch is compiled with only setup and loop.
in startup_stm32h750xbhx.s
running "bl __libc_init_array" causes WWDG_Handler before "main".

Program compiled using CubeIDE works fine. "main" can be called.
Program compiled using Arduino but part number in VBTx works fine on STM32H750XBH6. "main" can be called.

fpistm, 2024-11-05:
Hi @thisdp 
Thanks for the issue.
I think it is linked to the linker script. If you used the one from the VBTx it will works.
I have to analyze deeply to understand the root cause.

thisdp, 2024-11-05:
> Hi @thisdp Thanks for the issue. I think it is linked to the linker script. If you used the one from the VBTx it will works. I have to analyze deeply to understand the root cause.

VBTx IBKx can work, I haven't tested others, but I am sure XBHx can not work.

fpistm, 2024-11-05:
This is what I said if you used the linker script of the VB with the HB it will work.

thisdp, 2024-11-05:
> This is what I said if you used the linker script of the VB with the HB it will work.

I mean that I am not sure if other chips/part numbers have the same problem

closed 2024-11-07 by fpistm