stm32duino/Arduino_Core_STM32 #2557: docs: Add a new variant (board) and "Generic System Clock configuration"

thernstig, 2024-11-08:
I had a hard time understanding https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-%28board%29#3---generic-system-clock-configuration.

I am quoting the parts I found hard and what I felt was missing/strange:

> Go to Pinout tab, enable the desired peripherals which require specific clock configuration (not needed for peripherals clocked by HCLKx, or APBx clock): SDIO, USB, ... In this example only USB needs to be enabled as other peripherals default clock are correct by default.

What is "desired peripherals" here? How would I know what to enable?

> Set PLL Source Mux to HSI.

There is no `PLL Source Mux` for the MCU I added in https://github.com/stm32duino/Arduino_Core_STM32/pull/2556. The sentence is written like it is something that should exist, but it does not seem true for all MCUs.

thernstig, 2024-11-09:
In addition, this part:

> 6 - Restart
> Restart Arduino IDE and try one of the new entry with the [CheckVariant example](https://github.com/stm32duino/STM32Examples/tree/main/examples/NonReg/CheckVariant).

It mentions nothing on how to do this on that above page. Should there have been a link to somewhere on how to load the new board into Arduino to test it?

fpistm, 2024-11-09:
The board should appears in the menu you add it that's all. Select it and voilà.

thernstig, 2024-11-09:
@fpistm I am talking about an unreleased version of Arduino_Core_STM32. Then the new board does not immediately show up in the JSON file does it?

fpistm, 2024-11-09:
Yes. I need to perform a release.

thernstig, 2024-11-09:
@fpistm I think this part:

> But if you use the git repository in the Arduino package you will see it.

Is not explained from https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-%28board%29#6---restart.

How would one know how to do this? I just started with both Arduino and STM32, so consider it from a point from someone who never used these tools before.

fpistm, 2024-11-09:
It is explained here: https://github.com/stm32duino/Arduino_Core_STM32/wiki/Using-git-repository

Feel free to propose update, it is a community project.

thernstig, 2024-11-09:
I would propose a lot more if I knew the answers 😆

thernstig, 2024-11-09:
Wrote https://github.com/stm32duino/Arduino_Core_STM32/issues/2559 as a separate issue.

The two core issues of this topic are still valid. I do not know how to update those as I have no idea what those parts in reality mean.