stm32duino/Arduino_Core_STM32 #2559: docs: Improve wiki on how to verify new board variants

thernstig, 2024-11-09:
https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-%28board%29#6---restart does not explain how to verify the new changes.

thernstig, 2024-11-09:
@fpistm there is no way for anyone to update the wiki except for you. I tried to clone it and push here, but it did not work. Creating PRs for wikis is also not possible afaik.

So I did an update here you can check and update yourself in the Wiki of this repo:
https://github.com/thernstig/Arduino_Core_STM32/wiki/Add-a-new-variant-%28board%29#6---verify-your-changes

Compare with your original:
https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-%28board%29#6---restart

fpistm, 2024-11-12:
Hi  @thernstig 
Your changes has been merged:
https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-%28board%29#6---verify-your-changes

Thanks for this contribution.

closed 2024-11-12 by fpistm