stm32duino/Arduino_Core_STM32 #2577: STeaMi Board missing definition

SergioSuligoy, 2024-11-25:
STM32 Core 2.9.0

In boards.txt

In Line 13674 missing definition

STeaMi.build.board=STeaMi

fpistm, 2024-11-25:
Hi @SergioSuligoy 
it is defined later:
https://github.com/stm32duino/Arduino_Core_STM32/blob/7114677d971ab76d7c3ea19d3ebbb0840f0c715f/boards.txt#L13689

I agree other menu define it at top level but it is overridden later in any case.
Did you see an issue with that?

fpistm, 2024-11-25:
Even if we do not support Ardunio IDE 1.x legacy, I've tested and see that when STeami is not selected there is a warning:

`Warning: Board STMicroelectronics:stm32:STeaMi doesn't define a 'build.board' preference. Auto-set to: STM32_STEAMI` 

Anyway there is no issue to build and properties are properly set.