stm32duino/Arduino_Core_STM32 #2582: Bug: Missing USB CDC config for `GenC0`

thernstig, 2024-11-26:
When checking boards.txt I see things like:

```
GenG0.menu.usb.none=None
GenG0.menu.usb.CDCgen=CDC (generic 'Serial' supersede U(S)ART)
GenG0.menu.usb.CDCgen.build.enable_usb={build.usb_flags} -DUSBD_USE_CDC
GenG0.menu.usb.CDC=CDC (no generic 'Serial')
GenG0.menu.usb.CDC.build.enable_usb={build.usb_flags} -DUSBD_USE_CDC -DDISABLE_GENERIC_SERIALUSB
GenG0.menu.usb.HID=HID (keyboard and mouse)
GenG0.menu.usb.HID.build.enable_usb={build.usb_flags} -DUSBD_USE_HID_COMPOSITE
````

But I cannot see this for `GenC0`. Is this a bug/oversight?

fpistm, 2024-11-26:
Hi @thernstig 
USB on C0 is new. It cames with C071xx. 
I have to test it as I didn't have the board when added. 
I've received the board yesterday and so I will be able to test soon.

fpistm, 2024-11-26:
Note you can add it using Arduino feature:
 boards.local.txt

thernstig, 2024-11-26:
> Note you can add it using Arduino feature: boards.local.txt

I unfortunately do not know how to test it myself :(

fpistm, 2024-11-27:
> > Note you can add it using Arduino feature: boards.local.txt
> 
> I unfortunately do not know how to test it myself :(

You can refer to Arduino documentation: https://arduino.github.io/arduino-cli/0.34/platform-specification/#boardslocaltxt

> @fpistm I am not even sure how this works when using the STM32 MCUs such as C071xx. If we look at ATMega MCUs, their bootloader supports USB CDC by default and can be set to upload new firmware via the Arduino IDE via USB CDC.
> 
> But STM32 C071 default booatloader does not have USB CDC; only USB DFU.
> 
> So I cannot even grasp how it can work for this stm32duino to allow for the Arduino IDE to upload new firmware via the COM-port.

Arduino probably use the 1200bps feature which is catched by the USB CDC then reboot in bootloader mode.
At this time this is not supported here. You have to manually enter in BL mode.
I've made a PR several years ago as a POC but never find the time to end it (#710).

Anyway, I've tested USB and enable it properly, see #2584.

PR #2584 (2024-11-27): chore(c0): enable usb
Enable the HSI48 for USB (not done thanks CubeMX code generation). Add GenC0 USB menu.
Moved G0 one in alphabetical order.

Fixes #2582.

Tested with Nucleo C071RB.
commit 6f28a488 (2024-11-27): chore(c0): enable usb

Enable the HSI48 for USB (not done thanks CubeMX code generation).
Add GenC0 USB menu.
Moved G0 one in alphabetical order.

Fixes #2582.
closed 2024-11-27 by fpistm